node-hid/node-hid #424: Are there any prebuilds for Electron/linux-x64?

SteveJay7, 2021-04-18:
I am trying to build my electron app for multiple target platforms.
When running electron-builder on my Win10 machine, the win-x64 works fine but for linux-x64 it fails, ...

... getting a 404 for the expected prebuilt package "https://github.com/node-hid/node-hid/releases/download/v2.1.1/node-hid-v2.1.1-electron-v76-linux-x64.tar.gz"

... and saying "cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible"

so is the only way here to manually build a package on a linux machine and copying over to my Win10 machine to include it in the main building-process or are there any prebuilt packages for linux-64 / Electron available?

Thanks for your help!

todbot, 2021-04-18:
Are you using `electron-rebuild`?  `electron-rebuild` has a bug where it's passing both `--runtime=napi` and `--runtime=electron` to `prebuild-install`. See: https://github.com/electron/electron-rebuild/issues/520

Under NAPI, there is only one binary for both Node and Electron apps.

If not, it sounds like `electron-builder` is functioning in a similar broken way.  Perhaps the solution is to open an issue on `prebuild-install` to ignore any other `--runtime` if `--runtime-napi` is specified.

todbot, 2021-04-18:
I've opened a `prebuild-install` issue to attempt to solve this downstream.
 https://github.com/prebuild/prebuild-install/issues/146

SteveJay7, 2021-04-19:
Thanks for addressing that at prebuild-install.
I am using electron-builder, so it seems that behaviour is the same as in electron-rebuild.
Below is the whole output. What would be the best workaround? Maybe to download the linux-x64 tar.gz manually and rename it and put it into my npm-cache folder?

`  • electron-builder  version=22.5.1 os=10.0.19042
  • loaded configuration  file=package.json ("build" field)
  • writing effective config  file=build\builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=@serialport/bindings@9.0.7, node-hid@2.1.1 platform=linux arch=x64
  • install prebuilt binary  name=@serialport/bindings version=9.0.7 platform=linux arch=x64
  • install prebuilt binary  name=node-hid version=2.1.1 platform=linux arch=x64
  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 6.0.1
    prebuild-install info looking for cached prebuild @ C:\Users\stefa\AppData\Roaming\npm-cache\_prebuilds\dd8400-node-hid-v2.1.1-electron-v76-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/node-hid/node-hid/releases/download/v2.1.1/node-hid-v2.1.1-electron-v76-linux-x64.tar.gz
    prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v2.1.1/node-hid-v2.1.1-electron-v76-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=8.2.3 runtime=electron arch=x64 libc= platform=linux)

                                    command='C:\Program Files\nodejs\node.exe' 'C:\Users\stefa\OneDrive\Dokumente\BUSINESS\cleanit\app.cleanit.at\node_modules\prebuild-install\bin.js' --platform=linux --arch=x64 --target=8.2.3 --runtime=electron --verbose --force
                                    workingDir=C:\Users\stefa\OneDrive\Dokumente\BUSINESS\cleanit\app.cleanit.at\node_modules\node-hid
error Command failed with exit code 1.`

SteveJay7, 2021-04-19:
I just recognized that in the posted command in my last comment there is only "--runtime=electron" and not "--runtime=napi".
So maybe it is my fault that I should add the reference to napi somewhere, so when this command is generated it contains "napi". but not sure where to specify that.