node-hid/node-hid #425: Can't read from device unless support program run first

davecranwell, 2021-04-18:
I've got a pretty shonky USB device for telephone caller ID detection and it comes with support software with a UI to read data from the device. 

If I plug in the device and run the a rudimentary node scripts using `node-hid`, my script hangs.

```
var HID = require('node-hid');
var device = new HID.HID(50635,2);

device.read((err,data) => {
    console.log(data)
})
```

It will remain in a hung state till I run the support software, at which point it dumps a Buffer. Once I close the software and wait a little while, the device appears to go back into a different mode (indicated, but not explained, by LED lights) and then my script will again hang if run again. 

I appreciate there's little you can say about the device itself, but is there anything I'm doing wrong, generally speaking, in trying to interact with this device in this way? Is this behaviour something seen in other devices? I'm completely new to HID or USB programming in general so I'm fully expecting I'm doing something inadvisable.

todbot, 2021-04-18:
HI!
You are doing everything right.  It sounds like the device is getting sent a "send data" command from the support software.  This is pretty common in a lot of these kinds of devices.

Normally to reverse-engineer the protocol, one would use a hardware USB protocol sniffer while running the Windows software. These are devices are pretty expensive. I know some folks have had success with [Wireshark](https://wiki.wireshark.org/CaptureSetup/USB#Windows) and [USBPcap](https://desowin.org/usbpcap/).  I have no experience with that, but if you get it working and can send me the packet dump, I can help decode it.  If you do try this, I would recommend doing it on non-critical computer, as USBPcap ties into the USB drivers pretty deep and there's the possibility of losing USB mouse & keyboard access.

davecranwell, 2021-04-18:
The device comes with an SDK on a mini CD, which includes a .dll meant to be used with the code examples they also handily provide (but none in a language in which I'm fluent!). I can understand the rough flow of most of the code examples and they all depend on this .dll completely, which is problematic. 

I've been considering trying decompiling the dll because I suspect this "send data" command will be only the tip of the iceberg. Would sending you the dll be of any use?

mcuee, 2023-05-06:
I guess this is out of scope for the node-hid project and this can be closed.