node-hid/node-hid #427: Mac OS vendor defined usage page not recognized

adriencarbonaro, 2021-05-08:
Hi,

I use `node-hid` on Windows 10 + Mac OS 11 (Big Sur) to recognize a HID device. On Windows, everything is working fine but on Mac OS, `HID.devices()` only return the 2 standard usage pages, not my vendor defined usage page `0xFF00`.

I've seen information saying that keyboard devices are owned by OS for security reasons, I confirm my device is not keyboard or mouse, and the usage page `0xFF00` is vendor defined.

Here are more information about the device :

**HID Descriptor**

```
Descriptor Version Number:   0x0111
Country Code:   0
Descriptor Count:   1

Type:   0x22  (Report Descriptor)
Length (and contents):   148
Raw Descriptor (hex)    0000: 05 0C 09 01 A1 01 85 0A  15 00 25 01 75 01 95 01  
Raw Descriptor (hex)    0010: 09 E9 81 06 09 EA 81 06  09 CD 81 06 75 05 81 01  
Raw Descriptor (hex)    0020: C0 05 0B 09 05 A1 01 85  0B 75 01 09 20 81 22 09  
Raw Descriptor (hex)    0030: 2F 81 06 09 21 81 06 09  07 05 09 75 01 95 01 81  
Raw Descriptor (hex)    0040: 02 75 04 81 01 05 08 15  00 25 01 09 17 09 09 09  
Raw Descriptor (hex)    0050: 18 09 20 75 01 95 04 91  22 05 0B 15 00 25 01 09  
Raw Descriptor (hex)    0060: 9E 75 01 95 01 91 22 75  01 95 03 91 01 C0 06 AC  
Raw Descriptor (hex)    0070: FF 09 00 A1 FF 09 01 15  00 26 FF 00 75 08 85 0C  
Raw Descriptor (hex)    0080: 95 3F 81 22 09 02 15 00  26 FF 00 75 08 85 0D 95  
Raw Descriptor (hex)    0090: 3F 91 22 C0 

Parsed Report Descriptor:
Usage Page    (Consumer)
[...] (This usage page is found both Windows and MacOS)

Usage Page    (Telephony Device) 
[...] (This usage page is found both Windows and MacOS)

Usage Page (Vendor defined 0xFF00) (This usage page is not found on MacOS)
Usage 0 (0x0)
Collection (Collection )    
    Usage 1 (0x1)    
    Logical Minimum.........    (0)  
    Logical Maximum.........    (255)  
    Report Size.............    (8)  
    ReportID................    (0x0C)  
    Report Count............    (63)  
    Input...................    (Data, Variable, Absolute, No Wrap, Linear, No Preferred, No Null Position, Bitfield) 
    Usage 2 (0x2)    
    Logical Minimum.........    (0)  
    Logical Maximum.........    (255)  
    Report Size.............    (8)  
    ReportID................    (0x0D)  
    Report Count............    (63)  
    Output..................    (Data, Variable, Absolute, No Wrap, Linear, No Preferred, No Null Position, Nonvolatile, Bitfield) 
End Collection

Endpoint 0x83 - Interrupt Input
Address: 0x83  (IN)
Attributes: 0x03  (Interrupt)
Max Packet Size: 64
Polling Interval: 1 ms

Endpoint 0x02 - Interrupt Output
Address: 0x02  (OUT)
Attributes: 0x03  (Interrupt)
Max Packet Size: 64
Polling Interval: 1 ms
```

**`HID.devices()` return on MacOS**
```
{
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 12,
    usage: 1
  },
  {
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 11,
    usage: 5
  },
```

**`HID.devices()` return on Windows**
```
{
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 12,
    usage: 1
  },
  {
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 11,
    usage: 5
  },
  {
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 65280,
  },
```

Thank you for your help and for this lib.

todbot, 2021-05-08:
Since `node-hid` is just a thin wrapper on the C `hidapi` library, this question would be better answered there.

But it may be that your device, since it's advertising itself as a telephony-/audio-related device by C-Media, it could be there's a driver present on the Mac (either by the system or third-party) that is taking control of the device and not allowing access.  If you can, uninstall the driver on the Mac and try again.

closed 2021-05-08 by todbot

adriencarbonaro, 2021-05-08:
I just did. Thank you for your help. I've seen you're quite active on HIDAPI too, so sorry for the duplicate there.

I will try to see if a driver can be removed. I don't really know where to delete drivers on a mac. I will also try on another mac when I can.