node-hid/node-hid #429: Can not write big data (Buffer Length = 512) to device

ouxuwen, 2021-05-12:
node: v12.4.0
platform: win10
Hi,  I got problem when  trying to write data (Buffer 's length = 512) to my device, wirite data(Buffer's length = 64) is Ok, but write data (Buffer 's length = 512) get error "Cannot write to hid device" on windows 10.

On macos everything is ok.

Device:
Interface 4 HID Report Descriptor Vendor-Defined 1
Item Tag (Value) Raw Data 
Usage Page (Vendor-Defined 1) 06 00 FF  
Usage (Vendor-Defined 1) 0A 01 00  
Collection (Application) A1 01  
    Report ID (170) 85 AA  
    Logical Minimum (0) 15 00  
    Logical Maximum (-1) 25 FF  
    Report Size (8) 75 08  
    Report Count (63) 95 3F  
    Usage (Undefined) 09 00  
    Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) 81 02  
    Report Size (8) 75 08  
    Report Count (63) 95 3F  
    Usage (Undefined) 09 00  
    Output (Data,Var,Abs,NWrp,Lin,Pref,NNul,NVol,Bit) 91 02  
End Collection C0  
Usage Page (Vendor-Defined 2) 06 01 FF  
Usage (Vendor-Defined 1) 0A 01 00  
Collection (Application) A1 01  
    Report ID (187) 85 BB  
    Logical Minimum (0) 15 00  
    Logical Maximum (-1) 25 FF  
    Report Size (8) 75 08  
    Report Count (511) 96 FF 01  
    Usage (Undefined) 09 00  
    Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) 81 02  
    Report Size (8) 75 08  
    Report Count (511) 96 FF 01  
    Usage (Undefined) 09 00  
    Output (Data,Var,Abs,NWrp,Lin,Pref,NNul,NVol,Bit) 91 02  
End Collection C0

ert78gb, 2021-05-12:
Split the data to smaller chunks. We use this logic.

ouxuwen, 2021-05-26:
> Split the data to smaller chunks. We use this logic.

Use the bb channel, the Report Count is 512

mcuee, 2023-05-06:
@ouxuwen 

Interesting question. Have you tried latest HIDAPI using C (not node-hid) to see if it works under Windows 10 or not. Thanks.