node-hid/node-hid #431: HID over GATT

ks75vl, 2021-05-20:
I am working with a FIDO2 device, it is a HID device over GATT.

- On Window, look like we can manage that HID device, node-hid can open and communicate.

- On MacOS, look like we can't manage that HID device, but node-hid still can enumerate HID device, open device but can't write to that HID device. One more thing on MacOS, Webauthn client on Chrome can access those HID device and work fine.

Thanks for having a look!

todbot, 2021-05-20:
It sounds like Chrome is grabbing the device and opening it in exclusive mode.
Two questions:
- If you quit Chrome on MacOS, can `node-hid` access the device?
- What is the HID Report Descriptor for this device (you can use [mac-hid-dump](https://github.com/todbot/mac-hid-dump) to determine this)  
It may be there's an Apple-supplied driver for the FIDO2 usage page that prevents `node-hid` use (that is, the same as what happens with HID keyboards)

ks75vl, 2021-05-20:
HID Report Descriptor is below
```
06 d0 f1 09 01 a1 01 09 20 15 00 26 ff 00 75 08 95 40 81 02 09 21 15 00 26 ff 00 75 08 95 40 91 02 c0
```

- `node-hid` on MacOS can't access device in case Chrome quit (forced quit).
- I think Apple allow we access FIDO2 HID device, because I have another FIDO2 HID device (not HID over GATT), `node-hid` can access that device and work fine.

todbot, 2021-05-20:
Hmm. The usagePage in that HID Report Descriptor doesn't seem like it's a reserved one. 
Since`node-hid` is just a thin wrapper on the C-library [`hidapi`](https://github.com/libusb/hidapi/) and uses standard MacOS HID calls, perhaps there's someone else doing something similar using `hidapi`?

todbot, 2021-05-20:
Yes, sorry. I meant that the 0xF1D0 usagePage isn't one of the ones that MacOS will grab like Desktop, Keyboard, or Consumer.

ks75vl, 2021-05-20:
Yes
I trace down to `hidapi`, operator write to device failed in `IOHIDDeviceSetReport` with status code `kIOReturnUnsupported`