node-hid/node-hid #435: centos 8.x cannot open device with path /dev/hidrawl

gametheworld, 2021-07-28:
[image]

todbot, 2021-07-28:
Have you set udev rules for your device?

gametheworld, 2021-07-28:
> Have you set udev rules for your device?

no

gametheworld, 2021-07-28:
> Have you set udev rules for your device?
My current environment is running in the virtual machine vmware, I wonder if this will affect the opening of USB?

gametheworld, 2021-07-28:
> Have you set udev rules for your device?

i used the device = new HID.HID(vid,pid), Or an error
[image]

https://github.com/node-hid/node-hid#device--new-hidhidvidpid

gametheworld, 2021-07-28:
> Have you set udev rules for your device?

It worked on both Windows and MAC

todbot, 2021-07-28:
Please look up how to set up udev rules for USB devices.
https://github.com/node-hid/node-hid#udev-device-permissions
This is a common issue on Linux.

gametheworld, 2021-07-28:
> 请查看如何为 USB 设备设置 udev 规则。
> https://github.com/node-hid/node-hid#udev-device-permissions
> 这是 Linux 上的常见问题。

Thank you. Thank you. I just looked at the files and I found this,

closed 2021-07-28 by gametheworld

gametheworld, 2021-07-29:
@todbot My USB device needs a power-off and power-on process, but I made a mistake in teleporting after power-off, the same mistake as yesterday  ,May I ask why this is? Linux is not very well understood

todbot, 2021-07-29:
Hi @gametheworld, I'm not sure I understand.  If you mean you need to power on and off your device from Linux, there are tools in unix to turn on/off power to a port on the USB hub.  Here is a tool to do that: https://github.com/mvp/uhubctl

gametheworld, 2021-07-30:
> 嗨@gametheworld，我不确定我是否理解。如果你的意思是你需要从 Linux 上打开和关闭你的设备，unix 中有一些工具可以打开/关闭 USB 集线器端口的电源。这是一个工具：[https](https://github.com/mvp/uhubctl) : [//github.com/mvp/uhubctl](https://github.com/mvp/uhubctl)

Sorry, it is my firmware upgrade that causes the change of productid