node-hid/node-hid #439: Error: could not send feature report to device

zhangzhoufei, 2021-08-16:
**1. Question: Is the way I call the method to pass parameters wrong?**
 Error: could not send feature report to device
 
**2. My code:**
var HID = require('node-hid');
 this.HIDHID = new HID.HID(this.AppDevice.path);
 let sendFeatureData1 = this.HIDHID.sendFeatureReport([0x01,0x21,9,0x00,0x03,0x00,0x00,0x08,0x00]);
 let sendFeatureData2 = this.HIDHID.write([0xa1,0xf0,0x00,0x00,0x00,0x00,0x00,0x00]);
**3. This is my hardware device document:**
[http://tiebapic.baidu.com/forum/w%3D580/sign=d7a6dceec7160924dc25a213e406359b/f5693b385343fbf2d34a1d9ba77eca8064388fdd.jpg](url)
[http://tiebapic.baidu.com/forum/w%3D580/sign=598038754566d0167e199e20a72ad498/061fed5c103853439ba93c998413b07ecb8088dd.jpg](url)
**4. How should I use node-hid correctly to achieve it?**

todbot, 2021-08-16:
Hi,
It's difficult to tell what HID reports need to be sent by those screenshots.  Can you provide the HID Report Descriptor for this device?  If you're on Windows, you can get the HID Report Descriptor: https://github.com/todbot/win-hid-dump

zhangzhoufei, 2021-08-18:
> Hi,
> It's difficult to tell what HID reports need to be sent by those screenshots. Can you provide the HID Report Descriptor for this device? If you're on Windows, you can get the HID Report Descriptor: https://github.com/todbot/win-hid-dump
**[BLE Remote KMA Dongle] for my device found three win-hid-dump：**
1223:3D86: Telink - BLE Remote KMA Dongle
PATH:\\?\hid#vid_1223&pid_3d86&mi_02#8&386acad8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
1223:3D86: Telink - BLE Remote KMA Dongle
PATH:\\?\hid#vid_1223&pid_3d86&mi_04#8&120a925c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  A1  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  95  80  81  03  91  03  C1  00
  (31 bytes)
1223:3D86: Telink - BLE Remote KMA Dongle
PATH:\\?\hid#vid_1223&pid_3d86&mi_03#8&253aae9a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  01  09  01  A1  00  05  09  19  01
  29  05  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  05  81  02  95  03  81  03  05  01  09  30  25  7F  45  00
  75  08  95  01  81  06  09  31  81  06  09  38  81  06  C1  00
  C1  00
  (66 bytes)
  
 **How to use node-hid api afterwards？**

todbot, 2021-08-24:
Which of those 3 usages are you attempting to connect to?

todbot, 2021-08-24:
From using the[ USB Descriptor Parser](https://eleccelerator.com/usbdescreqparser/) as described on the `win-hid-dump` page, the first usage is keyboard, the second usage is vendor-defined (0xFF00), and the third usage is mouse.  Keyboard and mouse usages are claimed by the OS and `node-hid` cannot access them. 

The second usage is vendor-defined so you can access it, but you must open it specifically.  It defines an INPUT report and OUTPUT report, both with reportId of 0xA1 and a report count of 128 bytes. 

The code then would look something like this:  (this code is basically [test-teensyrawhid.js](https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js))

```js
var HID = require('node-hid')
var isTelinkBLE = function(d) { return d.vendorId===0x1223 && d.productId===0x3D86;}
var isRawHidUsage = function(d) { 
    return ((d.usagePage===0xFF00));
}
var reportId = 0xA1;
var reportLength = 128;

var devices = HID.devices();

var deviceInfo = devices.find( function(d) {
    return isTeensy(d) && isRawHidUsage(d);
});
if( deviceInfo ) { 
    device = new HID.HID( deviceInfo.path );
}
var report = new Array(reportLength + 1).fill(0);
report[0] = reportId;
report[1] = 0xF0;  // or whatever is the actual data to send
device.write(report);
```