node-hid/node-hid #442: Error: could not read from HID device during device.write()

alamathe1, 2021-08-24:
When writing to Jabra 410/510/710 speaker device, on macOS platform, we get the **Error** : `could not read from HID device`. (_NOTE_: This is **_not_** observed on Windows system, or on macOS system when it is connected via a dongle). 

Before `device.write()` is executed, we are able to successfully open a connection to device via `device = new HID.HID(path)`.

```
private send(data: number[]): void {

    try {
        this.device?.write(data);
    } catch (error) {
        logger.error('An error occurred writing to HID device', error);
    }
}
```

Is this a known issue? 

Also the reason that it only works on Windows and not on macOS when connected directly (as dongle connection works) - can indicate that there could be an underlying bug in the low level APIs?

todbot, 2021-08-24:
What kind of HID device is this?  What `usagePage` and `usage` are you connecting to? 

If you're not `open()`-ing on a specific usagePage/usage pair, and the device presents multiple usages in its HID Report Descriptor, then your app could be trying to open one owned by the OS (and if this is a telephony device, some usages will be claimed by the OS)

mcuee, 2023-05-06:
Since the OP has not come back for so long, I think this issue can be closed.