node-hid/node-hid #446: Installing node-hid: too few arguments to function call, expected 3, have 2

mmihic96, 2021-10-31:
Error while installing node-hid:

gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/HID.cc:612:31: error: too few arguments to function call, expected 3, have 2
npm ERR!   node::AtExit(deinitialize, 0);
npm ERR!   ~~~~~~~~~~~~                ^
npm ERR! /Users/milosmihic/Library/Caches/node-gyp/16.13.0/include/node/node.h:913:18: note: 'AtExit' declared here
npm ERR! NODE_EXTERN void AtExit(Environment* env,
npm ERR!                  ^
npm ERR! 1 error generated.
npm ERR! make: *** [Release/obj.target/HID/src/HID.o] Error 1

node version: 16.13.0
npm version: 8.1.0

todbot, 2021-11-01:
Could you give more details on how you're installing `node-hid`?  I just tried compiling clean with node 16.13.0 on MacOS 11.6.1 on Intel and it compiles.  I'm surprised you're not getting a prebuild. Normally compiling is not required.

mmihic96, 2021-11-01:
Hi @todbot,

I just do regular npm install node-hid. I have M1, maybe that is an issue?

todbot, 2021-11-01:
Ah good, because I just tested on an M1 Mac and all was good.
Looks like `node-hid-stream` is running a fairly old version of `node-hid`. 

Also, even if you get it installed, you may not like the results.  For security reasons, the OS prevents `node-hid` from accessing keyboards under most circumstances.

closed 2021-11-01 by todbot

mmihic96, 2021-11-03:
Hi,

I will just post here how I fixed it, maybe it helps somebody else in the future.
`npm shrinkwrap` resolved my issue. I changed `node-hid` version in their dependency.

https://nodejs.org/en/blog/npm/managing-node-js-dependencies-with-shrinkwrap/