node-hid/node-hid #447: TypeError: cannot open device with path IOService

lamfan, 2021-11-04:
I'm using macOS Monterey 12.0.1, node 16.13.0

var devices = HID.devices(); <-- it is working and return the device

  {
    vendorId: 4057,
    productId: 109,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS08@14300000/Stream Deck@14300000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice',
    serialNumber: 'AL16K2C60003',
    manufacturer: 'Elgato',
    product: 'Stream Deck',
    release: 512,
    interface: 0,
    usagePage: 12,
    usage: 1
  }

but when I create new HID.HID(device.path), it return error 

TypeError: cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14100000/USB2.0 Hub             @14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14140000/G913 TKL LIGHTSPEED Wireless RGB Mechanical Gaming Keyboard@14140000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice

Please help, many thanks!

todbot, 2021-11-04:
This sounds like there's another program running that's got the device open. Could that be the case?

lamfan, 2021-11-05:
> This sounds like there's another program running that's got the device open. Could that be the case?

I use HID.devices(); to get other hardware and create new HID.HID(device.path);

and I got the same results...

todbot, 2021-11-05:
I mean: do you have the Stream Deck software running on the same computer?  Or other software that controls the Stream Deck?

lamfan, 2021-11-05:
> I mean: do you have the Stream Deck software running on the same computer? Or other software that controls the Stream Deck?

Oh that's right, after I close the Stream Deck software, I can connect to it now.

So is there anyway to make both the Stream Deck software and node-hid work together?

todbot, 2021-11-05:
At the moment, no. Only one program can access a device at a time.  For HID devices normally this program is the OS, that then gives out events (like mouse & keyboard events) to other programs.

closed 2021-11-05 by todbot

lamfan, 2021-11-08:
> At the moment, no. Only one program can access a device at a time. For HID devices normally this program is the OS, that then gives out events (like mouse & keyboard events) to other programs.

Thank for your help todbot!