node-hid/node-hid #448: How to get HID report descriptor for Bluetooth

DrCWO, 2021-11-08:
I like to use node-hid for controling a Microsoft Surface Dial.
Has anyone an idea how to get the HID report descriptor?
The tools for USB give no result.

todbot, 2021-11-08:
Hi,  `nod-hid` cannot access devices that have been claimed by the OS or a custom device driver.  The Surface Dial is likely in this category.  You may be able to access the device under Linux or MacOS (where no driver exists) or by uninstalling the driver in Windows.

todbot, 2021-11-08:
And to specifically answer your question: To get the HID Report Descriptor for a device, use `usbhid-dump`: https://github.com/DIGImend/usbhid-dump  and then either `hidrd-convert` or the site https://eleccelerator.com/usbdescreqparser/

If the device is not found with `usbhid-dump` then it's not a HID device (according to Linux).

closed 2021-11-08 by todbot

DrCWO, 2021-11-08:
Hi Tod,
thank you for your quick response and I hope your suggestions is wrong.
**I don't see it with usbhid-dump but node-hid finds it and I can access it with node-hid.**

I'm currently working on a project on Linux (no desktop) with node.js.

Unfortunately I have a problem that I probably won't get solved without your knowledge.
I would therefore be really grateful for your support.

My hardware is a Microsoft Surface Dial that I access with node-hid.
I'm running it in RAW mode on a Raspberry Pi with Raspbian.

I get events for pressing and turning the dial without any problems and I can correctly decode them.
So the press and twist actions are reflected in HID. This works fine :-)

My challenge at the moment is controlling the haptic feedback.
In order to set everything up properly I have to use getFeatureReport and sendFeatureReport
and this is exactly where my problem begins:

I have no idea how to get the hid report descriptor of the dial (bluetooth).
In any case, the well-known USB tools do not provide it. Your experience would certainly help me here!
If I knew this, I would probably be easier for me to continue.

Due to a lack of information in this regard, I approached the topic using an example that I found here:
[https://github.com/daniel5151/surface-dial-linux/blob/main/src/dial_device/haptics.rs](https://github.com/daniel5151/surface-dial-linux/blob/main/src/dial_device/haptics.rs).

I started reading Report ID 1 with this command
               var r1 = hidSurfaceDial.getFeatureReport(1, 10);
and I get:
                [14,0,0,5,80,0]

Strangely enough, in order to be able to change values, I not only have to enter the report ID but also a 0x00 after it, like this:
               hidSurfaceDial.sendFeatureReport([0x01, 0x00, 0x03, 0x00, 0x00, 0x05, 0x50, 0x00]);

This changes the value 14 to 3. This change can be verified reading Report ID 1 again..
This change also causes the sensitivity of the dial to change, so everything seems to be good so far ;-)

However, I have the following questions to understand::
               - Why do I need the additional 0x00 in the 2nd array entry to make it write correctly?
               - Why the same values were written when the first bytes (Report ID) are not 0x01, 0x00 but 0x00, 0x00?

To set the waveform on the dial (buzz or click) I have to set features in Record ID 2. When I read it (length 32 bytes) I get this:
               [0,1,0,15,0,3,4,5,0]

Now I try to write the stuff back with a prefixed 0x02, 0x00::
                hidSurfaceDial.sendFeatureReport([0x02, 0x00, 0x00, 0x01, 0x00, 0x0f, 0x00, 0x03, 0x04, 0x05, 0x00]);

As a result, I see the following error message and now I'm stuck:
hidSurfaceDial.sendFeatureReport([0x02, 0x00, 0x00, 0x01, 0x00, 0x0f, 0x00, 0x03, 0x04, 0x05, 0x00]);
              ^
TypeError: could not send feature report to device
    at Timeout.start [as _onTimeout] (/home/pi/rooExtend/HID_Test.js:57:17)
    at listOnTimeout (node:internal/timers:557:17)
    at processTimers (node:internal/timers:500:7)

I would be incredibly grateful for your support on two points:
               - How or where can I get the hid report descriptor of the Surface Dial (Bluetooth)?
               - How do I get the 2nd feature written?

Thank you for your support
Best Charly

[HID_Test.zip](https://github.com/node-hid/node-hid/files/7498702/HID_Test.zip)
y.github.com>
Cc: DrCWO <info@definiteaudio.de>; Author <author@noreply.github.com>

todbot, 2021-11-08:
Hi,
I see now that `usbhid-dump` is using `libusb` instead of `hidraw`.  That means it cannot pick up BLE devices like the Surface Dial.  Sorry about recommending it.

To hopefully help, I created a small tool called `hidraw-dump` that works sorta similarly.
You can download it here: https://github.com/todbot/hidraw-dump

It should at least let you dump the HID Report Descriptor of the Surface Dial so you can figure out the type of reports it expects  It is critical to get the following correct:
- type of report (FEATURE or OUTPUT when sending to device)
- reportId of report (looks like "0x01" here) 
- the size in bytes to send of the report ("REPORT_COUNT" in HID Report Descriptor lingo)

From that `haptics.rs` you link, it appears that the `DialHIDWrapper::buzz()` method does:
- OUTPUT report
- reportId 0x01
- report size is 4 bytes long (5 with reportId)

I think. I don't actually know Rust.  In NodeJs using `node-hid`, I think that would look like:

```js
var reportCount = 3;
var buzzReport = [ 0x01, reportCount, 0x03, 0x00, 0x00 ];
hidSurfaceDial.write( buzzReport );
```
Note that `write()` is used instead of `sendFeatureReport()` because this appears to be an OUTPUT report instead of a FEATURE report.

DrCWO, 2021-11-09:
Hi Tod,

thank you so much. I posted the record descriptor in the Issue-List of hidraw-dump.

 

First had some trouble installing it but I closed this issue now as I simply forgot to upgrade my Raspbian first.

 

I will dig into it during the next days and see how I can achieve to get a “click” instead of a “buzz”.

I understand what HID is for but it also seems overcomplicated for simple Use Cases like creating a “click” ;-)

 

Thanks’ again and I’l get you informed regarding my Surface Dial work.

 

Best Charly