node-hid/node-hid #449: error: `ISO C++17 does not allow dynamic exception specifications`

DanielAtKrypton, 2021-11-13:
# The problem
Compiling node-hid fails in my system...

# System
- Debian 11
- Node v17.1.0
- Node-gyp v8.3.0
- Windows System Linux 5.10.60.1-microsoft-standard-WSL2

# Expected results
Flawless compilation when adding `husky@7.0.4`

# Actual Results
```bash
  CC(target) Release/obj.target/hidapi-linux-hidraw/hidapi/linux/hid.o
  AR(target) Release/obj.target/hidapi-linux-hidraw.a
  COPY Release/hidapi-linux-hidraw.a
  CXX(target) Release/obj.target/HID-hidraw/src/HID.o
../src/HID.cc:67:5: error: ISO C++17 does not allow dynamic exception specifications
   67 |     throw(JSException);
      |     ^~~~~
../src/HID.cc:70:5: error: ISO C++17 does not allow dynamic exception specifications
   70 |     throw(JSException);
      |     ^~~~~
../src/HID.cc:152:3: error: ISO C++17 does not allow dynamic exception specifications
  152 |   throw(JSException)
      |   ^~~~~
../src/HID.cc:163:3: error: ISO C++17 does not allow dynamic exception specifications
  163 |   throw(JSException)
      |   ^~~~~
../src/HID.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE HID::read(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/HID.cc:255:52: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  255 |   uv_queue_work(uv_default_loop(), req, recvAsync, (uv_after_work_cb)recvAsyncDone);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [HID-hidraw.target.mk:124: Release/obj.target/HID-hidraw/src/HID.o] Error 1
```

todbot, 2021-11-13:
What version of `node-hid` are you trying to compile?

Regardless, I am pretty sure you cannot compile native Node modules in WSL because they think they're in Linux when they need the access to Windows shared libs/APIs.

What is ` husky@7.0.4`? I'm not aware of it or how it would be relevant to `node-hid`.

DanielAtKrypton, 2021-11-14:
> What version of `node-hid` are you trying to compile?
>
> Regardless, I am pretty sure you cannot compile native Node modules in WSL because they think they're in Linux when they need the access to [...]

Native node modules are already being compiled in my environment.
This is Husky: https://github.com/typicode/husky

WSL2 is different than WSL1 in the sense that the former relies on Hyper-V virtualization engine to run Linux in an isolated container. See more about this here: https://www.ionos.co.uk/digitalguide/server/know-how/wsl2-explained/#:~:text=WSL2%2C%20which%20was%20released%20in%20early%202020%2C%20is,is%20developed%20and%20distributed%20as%20a%20commercial%20product.

From my perspective everything just runs and compiles fine in Debian 11 inside WSL2 other than `node-hid`, unfortunately.

DanielAtKrypton, 2021-11-14:
yarn is trying to compile node-hid@2.1.1

todbot, 2021-11-14:
Are you compiling any other native Node modules besides `node-hid`? 

If you notice in your results output above, `node-hid` is assuming it's in a Linux environment, so is trying to compile using the Linux hidraw APIs. If WSL2 does not provide hidraw, then you will not be able to compile `node-hid` in this environment. Please compile in standard Windows environment.

ryanswilson59, 2022-06-07:
I get this same error when running `npm i` to build nault. I run manjaro linux 5.15.41-1

todbot, 2022-06-07:
> I get this same error when running `npm i` to build nault. I run manjaro linux 5.15.41-1

That sounds like a problem with `nault`.  It looks like it's using a very old version of `node-hid`:
https://github.com/Nault/Nault/blob/master/package.json#L71

closed 2022-06-07 by todbot

atillabirer, 2023-03-18:
this is a persisting issue on the latest node-hid, Ubuntu 22.04.

Julusian, 2023-03-18:
> this is a persisting issue on the latest node-hid, Ubuntu 22.04.

Do you mean the warnings or build failure?

atillabirer, 2023-03-18:
build failure, with the same error as title.

mcuee, 2023-05-06:
@atillabirer's issue on Ubuntu 22.04 is a different issue.
* https://github.com/node-hid/node-hid/issues/500

And the original issue reported is an invalid issue as WSL2 does not provide hidraw.

Maybe the issue should be closed.