node-hid/node-hid #450: ETA on Electron 12 support

wobedi, 2021-11-18:
Hi guys and gals,

Is there an ETA on when you'll support Electron 12?

And, as it stands today, to do you think that using `node-hid` with Electron 12 will:
a) Not work at all
b) Most likely not work
c) Most likely work

(I understand that this is dependent on upstream dependencies and out of your control, but I figured I'd ask, maybe you have some estimate :))

Reasoning: I need min. Electron 12.2.2 because of this bugfix:
https://github.com/ranmd9a/beatlist/pull/1

todbot, 2021-11-19:
I'm unclear what you are asking.  `node-hid` works fine in Electron v12 (v14, etc).  `node-hid` doesn't have Node or Electron version-based restrictions now.  You can see an example of an Electron app that uses `node-hid` here: https://github.com/todbot/electron-hid-toy

If you have more details about an error you're seeing, please include them. Closing this issue for now.

closed 2021-11-19 by todbot

wobedi, 2021-11-19:
I was basing my question on the fact that the very first section of your [README](https://github.com/node-hid/node-hid#platform-support) states:

> Supported Electron versions
> Electron v3 to
> Electron v11
> Any newer version of Electron or Node MAY NOT WORK

Which very much makes it sound like any Electron version beyond 11 is not supported and mightwill encounter issues?

If that is not the case, and Electron 12 and even 14 work fine, why not update the README?

todbot, 2021-11-19:
Simply because I don't not have infinite time to keep updating this package when this week's version comes out. :)

todbot, 2021-11-19:
And thank you for reminding me that the README should've been updated. Sorry if I was a bit curt before.

wobedi, 2021-11-22:
All good, I appreciate the massive amount of work that you are putting in to give us all superpowers with this library :) Thanks for updating the README!