node-hid/node-hid #453: Cannot open device: TypeError: cannot open device with path 0001:0002:00

jmorris644, 2021-12-13:
I am running node-hid on a raspberry pi with the most current versions of node.js and npm. The output from lsusb is:
```
Bus 001 Device 002: ID 0fd9:0080 Elgato Systems GmbH Stream Deck MK.2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
The output from HID.devices is:
```
vendorId: 4057
productId: 128
path: "0001:0002:00"
release: 512
interface: 0
```
I have tried both the path and the vid,pid. Both are getting the same error.

todbot, 2021-12-13:
Have you set [Linux udev device permissions](https://github.com/node-hid/node-hid#linux-notes)?

jmorris644, 2021-12-13:
Yes, I have set the device permissions. I tried both libusb and hidraw methods. I had put the error in the Title. It is:
```
TypeError: cannot open device with path 0001:0002:00
```
for type of libusb and:

```
TypeError: cannot open device with path /dev/hidraw0
```
for type of hidraw

I also tried opening with the vid,pid but that did not work either.

todbot, 2021-12-13:
A few more questions:
- Can you provide the code you are using?
- Are you running the latest version of node-hid?
- What does `HID.devices()` print out?
- Have you tried unplugging & plugging the device in? (This assures udev rules get run)
- Is there another app that has that device open? 
- Which `usage` and `usagePage` are you attempting to open on the device?  If the `usagePage` is an OS-owned one like mouse & keyboard, then you will not be able to access it.

jmorris644, 2021-12-14:
> A few more questions:
> 
> *1 Can you provide the code you are using?
> *2 Are you running the latest version of node-hid?
> *3 What does `HID.devices()` print out?
> *4 Have you tried unplugging & plugging the device in? (This assures udev rules get run)
> * 5Is there another app that has that device open?
> * 6Which `usage` and `usagePage` are you attempting to open on the device?  If the `usagePage` is an OS-owned one like mouse & keyboard, then you will not be able to access it.

1. I am using node-red.
```
var require = global.get('require');

var HID = require('node-hid');
HID.setDriverType('libusb');
//HID.setDriverType('hidraw');
var devices = HID.devices();
var device = new HID.HID(0x0fd9,0x0080);
//var device = new HID.HID(devices[0].path);
//var device = new HID.HID("/dev/hidraw0");
msg.payload = devices;
return msg;
```

2. 2.1.1
3. Single element array containing:
```
vendorId: 4057
productId: 128
path: "0001:0002:00"
release: 512
interface: 0
```
4. Yes, I have also rebooted completely
5. This is the only app. It is a brand new pi for this purpose only
6. Sorry, I do not understand this.

Additional info:

lsusb output
```
Bus 001 Device 002: ID 0fd9:0080 Elgato Systems GmbH Stream Deck MK.2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

todbot, 2021-12-14:
Why are you setting the driver type to `libusb`? What does `HID.devices()` print out when you remove `HID.setDriverType()`?

jmorris644, 2021-12-14:
Whether I set the driver to either option or I do not set it at all the HID.devices() does not change. It is always:
```
vendorId: 4057
productId: 128
path: "0001:0002:00"
release: 512
interface: 0
```

jmorris644, 2021-12-14:
If it helps when I run the show-devices.js in the src dir I get:
```
devices: [
  {
    vendorId: 4057,
    productId: 128,
    path: '/dev/hidraw0',
    serialNumber: 'DL38K1A04998',
    manufacturer: 'Elgato',
    product: 'Stream Deck MK.2',
    release: 512,
    interface: 0,
    usagePage: 12,
    usage: 1
  }
]
```

jmorris644, 2021-12-14:
I made progress. I modified the test-blink1.js to hold the appropriate VENDOR_ID and PRODUCT_ID. When I ran it I got the same error. I then ran it using sudo and it worked!

So, now what do I do so that I do not need to run node-hid with privileges?

todbot, 2021-12-14:
Normally installing udev permissions allows you to run as non-root.  Because this device is usagePage 12 ("Consumer Page", the same usagePage that media control keys like vol up/down are on), it is likely there is an OS-level driver for that device that has claimed it.

It looks like you're trying to use a Stream Deck. I don't have one so can't test, but there is an existing library that handles it. It may have some tips: https://www.npmjs.com/package/elgato-stream-deck

closed 2021-12-14 by todbot

Julusian, 2021-12-14:
@jmorris644
Having to chmod /dev/hidraw0 to make it work suggests that the udev rules are not setup correctly. 
If you have taken the udev rules from https://github.com/Julusian/node-elgato-stream-deck#linux then that could be because they do not include rules for hidraw mode, but they do work for libusb.

It really should work in libusb mode, I have a MK2 streamdeck that does work

> So I tried to install node-hib@0.5.7 as documented in the above issue and got tons of errors

I don't remember the version from memory, but I remember adding some functionality to the node-hid api that the streamdeck library relies upon, so going backwards very far before node-hid@1.0.0 is unlikely to work.

>  Is the --driver option still relevant?

I don't think so. in recent versions of node-hid both libusb and hidraw modes are always built, with it chosen at runtime bsaed on `HID.setDriverType('libusb');`

jmorris644, 2021-12-14:
So I went back to the basics:

I set up a test directory and installed node-hib.

I then created a file called streamdeck.rules in /etc/udev/rules.d with the content:
```
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0fd9", ATTRS{idProduct}=="0080", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="0fd9", ATTRS{idProduct}=="0080", MODE="0666", GROUP="plugdev"
```
I then ran sudo udevadm control --reload-rules

I then ran the show-devices.js and it resulted in:
```
devices: [
  {
    vendorId: 4057,
    productId: 128,
    path: '/dev/hidraw0',
    serialNumber: 'DL38K1A04998',
    manufacturer: 'Elgato',
    product: 'Stream Deck MK.2',
    release: 512,
    interface: 0,
    usagePage: 12,
    usage: 1
  }
]
```
I then added a HID.setDriverType('libusb') and got:
```
devices: [
  {
    vendorId: 4057,
    productId: 128,
    path: '0001:0005:00',
    release: 512,
    interface: 0
  }
]
```
I then added hidDevice = new HID.HID(0x0fd9,0x0080) at the end of show-devices.js and got:
```
devices: [
  {
    vendorId: 4057,
    productId: 128,
    path: '0001:0005:00',
    release: 512,
    interface: 0
  }
]
/home/joe/test/node_modules/node-hid/nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

TypeError: cannot open device with vendor id 0xfd9 and product id 0x80
    at new HID (/home/joe/test/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/home/joe/test/node_modules/node-hid/src/show-devices.js:16:13)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12)
    at internal/main/run_main_module.js:17:47
```
I don't know what else to do.

Julusian, 2022-01-24:
@jmorris644 
> I then ran sudo udevadm control --reload-rules

Did you then unplug and replug the device?

jmorris644, 2022-01-24:
It has been so long now I do not remember how I finally got it all running. If I remember, correctly I modified the package.json and updated a version of a dependency. But don't hold me to that. Sorry.