node-hid/node-hid #455: When compiling source (Ubuntu:Arm8) cannot install gcc-4.8+:

jmorris644, 2021-12-15:
When executing the commands: ```
apt install gcc-4.8 g++-4.8 && export CXX=g++-4.8
```
I get:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 g++-4.8 : Depends: libmpfr4 (>= 3.1.3) but it is not installable
 gcc-4.8 : Depends: cpp-4.8 (= 4.8.5-4ubuntu8~16.04.1) but it is not going to be installed
           Depends: libmpfr4 (>= 3.1.3) but it is not installable
```

When I try to install libmpfr4 I get:
```
Package libmpfr4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
```
searching for a solution has been unsuccessful.

todbot, 2021-12-16:
Why are you trying to install a specific version of gcc?
Generally when I'm developing on Ubuntu I do:
```
sudo apt install build-essential
```

jmorris644, 2021-12-16:
Yes, I thought that too. So then I went back to the install instructions and noted the prerequisites:
```
Linux (kernel 2.6+) : (install examples shown for Debian/Ubuntu)
Compilation tools: apt install build-essential git
gcc-4.8+: apt install gcc-4.8 g++-4.8 && export CXX=g++-4.8
libusb-1.0-0 w/headers:apt install libusb-1.0-0 libusb-1.0-0-dev
libudev-dev: apt install libudev-dev (Debian/Ubuntu) / yum install libusbx-devel (Fedora)
```

WHen I did this gcc-4.8+ would not install because of missing libmpft4.

But, if I back up a little bit, my original error when building is:
```
 npm install node-hid --build-from-source

> node-hid@2.1.1 install /home/joe/test/node_modules/node-hid
> prebuild-install --runtime napi || node-gyp rebuild

/bin/sh: 1: pkg-config: not found
gyp: Call to 'pkg-config libusb-1.0 --cflags-only-I | sed s/-I//g' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:282:12)
gyp ERR! System Linux 5.4.0-1047-raspi
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/joe/test/node_modules/node-hid
gyp ERR! node -v v14.18.2
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/joe/test/package.json'
npm WARN test No description
npm WARN test No repository field.
npm WARN test No README data
npm WARN test No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@2.1.1 install: `prebuild-install --runtime napi || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@2.1.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```
I am not super experienced with stuff at this level. Should I be looking somewhere else to resolve the issue?

todbot, 2021-12-16:
Looks like you're missing `pkg-config`. Try:
```
sudo apt install pkg-config
```
And then recompiling.

todbot, 2021-12-16:
This is a deficiency in the README, not your fault. We accepted a PR a while back that introduced pkg-config as a dependency and I didn't update the README. Apologies for that. 
Also, the build and error messages from `node-gyp` are pretty cryptic. It's not surprising you missed it.

jmorris644, 2021-12-16:
Thanks. And I see that you modded the docs.
When Id di the install I received these warnings but it looks like it finished the build
```
make: Leaving directory '/home/joe/test/node_modules/node-hid/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/joe/test/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/joe/test/package.json'
npm WARN test No description
npm WARN test No repository field.
npm WARN test No README data
npm WARN test No license field.
```

todbot, 2021-12-16:
That's the warning you get when installing a NodeJs package locally in a directory that's not part of a project. A NodeJs project normally has a `package.json` file and when you do `npm install --save <packagename>`, npm writes the packagename and version into the `dependencies` section of the `package.json` file.

A lot of people when starting out a NodeJs project will do `npm init -y` to create a starter package.json file.

jmorris644, 2021-12-16:
Makes sense. I am closing this issue as the docs were updated so it will not occur again.

closed 2021-12-16 by jmorris644