node-hid/node-hid #459: Feature Request: use nodeGypBuild and prebuildify

Lan-Hekary, 2022-01-24:
Hello, 
I am using the library with electron I am using other native libraries too.
All the others are moving toward NAPI and you successfully made that transition.

Also they are using [nodGypBuild](https://www.npmjs.com/package/node-gyp-build) to load the bindings instead of 'bindings' library.

the advantage is that using it with [prebuildify](https://github.com/prebuild/prebuildify) eliminate the need for rebuilding the native module for every operating system individually and include all of the prebuilt libraries in the output directly.

Please Check it out and tell me your thoughts on the matter.

todbot, 2022-01-24:
Unfortunately `nodeGypBuild` and `prebuildify` do not support multiple `.node` files, as needed by the Linux build of `node-hid`. For more details see: https://github.com/prebuild/prebuildify/issues/57

Leaving this issue open for now as a reminder to look back into this periodically.

Julusian, 2025-06-03:
A while back this library changed to using a library that works similarly to node-gyp-build but that works with bundlers like webpack better.
So I think this is handled

closed 2025-06-03 by Julusian