node-hid/node-hid #460: device unresponsive while reading

clonq, 2022-01-29:
The physical device becomes unresponsive while listening to, or reading from it.
A simple `device.on("data", function(data) { console.log(data) });` seems to freeze the physical device.
Data gets captured properly but there is no visual feedback on the device while interacting with it (e.g pressing a button should turn on a led)
I've tried both `device.on` and `device.getFeatureReport` within the main app and in a worker thread, same result.
Any ideas what I'm doing wrong?

todbot, 2022-01-29:
Without knowing anything about your device I can offer no assistance sorry.

clonq, 2022-01-29:
Sorry @todbot, it's a Native Instruments Traktor Kontrol S4 MK3

todbot, 2022-01-29:
Do you have the USB Config Descriptor and the USB HID Report Descriptor for this device?

clonq, 2022-01-29:
Not really. The manufacturer doesn't seem keen to make that public. I discovered by trial an error a few bytes that I'm experimenting with. I know for example that the 3rd byte from the 1st report stores a bit-masked state for 4 buttons. And that's enough for me to test a proof of concept.

clonq, 2022-01-31:
@todbot does that help? Is there anything you could suggest?

todbot, 2022-01-31:
Without getting the USB Config Descriptor or USB HID Report Descriptor, there's not much I can do.

todbot, 2022-01-31:
You can get the HID Report Descriptor (the more important of the two), using a "hid-dump" tool.  
https://todbot.com/blog/2021/01/29/get-hid-report-descriptors-with-win-hid-dump-mac-hid-dump/

clonq, 2022-01-31:
Ok, I'll check the link, Thanks!

clonq, 2022-01-31:
[image]
Any idea why I get this?

clonq, 2022-01-31:
Hi @todbot - I built it from source and ran it. Here's the descriptor:

`
06  01  ff  09  00  a1  01  09  01  a1  02  85  01  09  02  15
00  25  01  75  01  95  88  81  02  09  07  15  00  25  01  75
01  95  10  81  02  09  03  15  00  25  0f  75  04  95  06  81
02  c0  09  01  a1  02  85  02  09  05  15  00  26  ff  0f  75
10  95  07  81  02  09  04  15  00  26  ff  0f  75  10  95  20
81  02  c0  09  30  a1  02  85  03  09  31  15  00  26  ff  00
75  08  95  03  81  02  09  31  15  00  26  ff  00  75  08  95
1c  81  02  09  31  15  00  26  ff  00  75  08  95  1c  81  02
c0  09  01  a1  02  85  04  09  31  15  00  26  ff  00  75  08
95  27  81  02  c0  09  30  a1  02  85  30  09  31  15  00  26
ff  00  75  08  95  1b  91  02  c0  09  30  a1  02  85  31  09
f4  15  00  26  ff  00  75  08  95  0a  91  02  c0  09  30  a1
02  85  32  09  81  15  00  26  ff  00  75  08  95  28  91  02
c0  09  80  a1  02  85  80  09  81  15  00  25  7f  75  08  95
5e  91  02  c0  09  80  a1  02  85  81  09  81  15  00  25  7f
75  08  95  4e  91  02  c0  09  80  a1  02  85  82  09  81  15
00  25  7f  75  08  95  08  91  02  c0  09  80  a1  02  85  8e
09  00  15  00  26  ff  00  75  08  95  3e  91  02  c0  09  d8
a1  02  85  d8  09  d1  15  00  27  ff  ff  00  00  75  10  95
02  b1  03  09  d1  15  00  27  ff  ff  00  00  75  10  95  02
b1  03  09  d1  15  00  27  ff  ff  00  00  75  10  95  04  b1
03  09  d1  15  00  26  ff  00  75  08  95  01  b1  03  09  d1
15  00  26  ff  00  75  08  95  0f  b1  03  c0  09  d8  a1  02
85  d9  09  d1  15  00  26  ff  00  75  08  95  20  b1  03  c0
09  f0  a1  02  85  f2  09  e6  15  00  25  64  75  08  95  03
b1  02  09  e6  15  00  25  64  75  08  95  03  b1  02  c0  09
f0  a1  02  85  f4  09  f4  15  00  26  ff  00  75  08  95  01
91  02  09  f5  15  00  26  ff  00  75  08  95  1f  91  02  c0
09  e0  a1  02  85  f8  09  e1  15  00  27  ff  ff  00  00  75
10  95  01  b1  03  09  e2  15  00  27  ff  ff  00  00  75  10
95  01  b1  03  09  e3  15  00  26  ff  00  75  08  95  01  b1
03  09  e4  15  00  26  ff  00  75  08  95  01  b1  03  09  e5
15  00  26  ff  00  75  08  95  01  b1  03  09  e7  15  00  25
64  75  08  95  01  b1  02  09  e6  15  00  25  64  75  08  95
01  b1  02  09  e8  15  00  25  01  75  08  95  01  b1  02  c0
09  e0  a1  02  85  f9  09  e1  15  00  27  ff  ff  00  00  75
10  95  01  b1  03  09  e2  15  00  27  ff  ff  00  00  75  10
95  01  b1  03  09  e3  15  00  26  ff  00  75  08  95  01  b1
03  09  e4  15  00  26  ff  00  75  08  95  01  b1  03  09  e5
15  00  26  ff  00  75  08  95  01  b1  03  09  e7  15  00  25
64  75  08  95  01  b1  02  09  e6  15  00  25  64  75  08  95
01  b1  02  09  e8  15  00  25  01  75  08  95  01  b1  02  c0
c0
`

todbot, 2022-01-31:
From https://eleccelerator.com/usbdescreqparser/ the parsed output is:

```
0x06, 0x01, 0xFF,  // Usage Page (Vendor Defined 0xFF01)
0x09, 0x00,        // Usage (0x00)
0xA1, 0x01,        // Collection (Application)
0x09, 0x01,        //   Usage (0x01)
0xA1, 0x02,        //   Collection (Logical)
0x85, 0x01,        //     Report ID (1)
0x09, 0x02,        //     Usage (0x02)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x88,        //     Report Count (-120)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x07,        //     Usage (0x07)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x10,        //     Report Count (16)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x03,        //     Usage (0x03)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x0F,        //     Logical Maximum (15)
0x75, 0x04,        //     Report Size (4)
0x95, 0x06,        //     Report Count (6)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x09, 0x01,        //   Usage (0x01)
0xA1, 0x02,        //   Collection (Logical)
0x85, 0x02,        //     Report ID (2)
0x09, 0x05,        //     Usage (0x05)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x0F,  //     Logical Maximum (4095)
0x75, 0x10,        //     Report Size (16)
0x95, 0x07,        //     Report Count (7)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x04,        //     Usage (0x04)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x0F,  //     Logical Maximum (4095)
0x75, 0x10,        //     Report Size (16)
0x95, 0x20,        //     Report Count (32)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x09, 0x30,        //   Usage (0x30)
0xA1, 0x02,        //   Collection (Logical)
0x85, 0x03,        //     Report ID (3)
0x09, 0x31,        //     Usage (0x31)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x03,        //     Report Count (3)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x31,        //     Usage (0x31)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x1C,        //     Report Count (28)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x31,        //     Usage (0x31)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x1C,        //     Report Count (28)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x09, 0x01,        //   Usage (0x01)
0xA1, 0x02,        //   Collection (Logical)
0x85, 0x04,        //     Report ID (4)
0x09, 0x31,        //     Usage (0x31)
[... 246 lines cut ...]
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0xE8,        //     Usage (0xE8)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xC0,              // End Collection

// 673 bytes
```

These are all in the vendor-defined usagePage area, so we can get no help from standard HID usage tables. 

There are also several different usages defined.  Normally when you open a device with multiple usages, you open up the specific usage you want.  There's also multiple reports per usage, some defined as INPUT reports (to the computer), OUTPUT reports (to the device), and FEATURE reports (can be to or from computer).  

Unfortunately, this device has a very complicated set of data packets (reports) it can send and receive.  Without knowing what those different reports are for and how they're constructed, I'm afraid there's not much to do.

The reason why you got that alert is because the app isn't signed. If you right-click on it (or run it from the Terminal), it should work.

It looks like this app supports the Traktor Kontrol S4: https://github.com/mixxxdj/mixxx 
perhaps you can glean some details from it

clonq, 2022-01-31:
@todbot thank you for the time you put into this. I'm still confused though. Say I manage to somehow figure out the usages and the reports. How would that help to answer my original question: why the device becomes unresponsive?

todbot, 2022-01-31:
Without knowing the data spec, who knows? It may be waiting for some response from the computer and never getting it.

clonq, 2022-02-02:
Hi @todbot. It looks like as soon as I initialize the HID device with `new HID.HID(device.path);`, the device becomes unresponsive. It's like my code gets a lock on the device. Have you seen anything like this?

todbot, 2022-02-02:
Every device is different.  What do you mean by "unresponsive"?  If you mean you have another program trying to access the device and that stops working when you use `node-hid`, then yes that is expected behavior: only one program can access the device at a time.

clonq, 2022-02-02:
By unresponsive I mean that the hardware doesn't give any feedback. If I push a button, the associated LED won't turn on.

todbot, 2022-02-03:
Without knowing how the HID Report API of the device, it's impossible to respond if this is expected behavior or not. It may be that when the host connects to the device, the device may require responses from the computer to act normally.

closed 2022-02-03 by todbot