node-hid/node-hid #462: A crash issue in electron(dump files attached)

Vict0r-Chen, 2022-02-14:
## Environment
**OS**: win10 21H2
**nodejs**: v12.22.1
**electron**: v13.1.2
**node-hid**: v2.1.1
**ffi-napi**: v3.1.0

## About
Sorry, it's a little difficult to explain this in detail.
In general, I call the C++ class library through ffi-napi in Electron, and then communicate with the device through node-hid in the callback function of the C++ class library.
In this scenario, some computers will crash, the probability is about 5%, and some computers may be less than 1%. I uploaded several dump files, some of which are below.
```
STACK_TEXT:  
0000005f`8bbfb6b0 00007ffa`fc1d2bf4     : 00003f00`015c5320 00003f00`013c2480 00000000`00000000 00007ff7`97e86a3c : KERNELBASE!RaiseException+0x69
0000005f`8bbfb790 00003f00`015c5320     : 00003f00`013c2480 00000000`00000000 00007ff7`97e86a3c 00007ffa`fc170000 : HID+0x62bf4
0000005f`8bbfb798 00003f00`013c2480     : 00000000`00000000 00007ff7`97e86a3c 00007ffa`fc170000 00000000`19930520 : 0x00003f00`015c5320
0000005f`8bbfb7a0 00000000`00000000     : 00007ff7`97e86a3c 00007ffa`fc170000 00000000`19930520 0000005f`8bbfb830 : 0x00003f00`013c2480
STACK_COMMAND:  ~0s; .ecxr ; kb
SYMBOL_NAME:  hid+62bf4
MODULE_NAME: HID
IMAGE_NAME:  HID.node
FAILURE_BUCKET_ID:  CPP_EXCEPTION_e06d7363_HID.node!Unknown
```
```
STACK_TEXT:  
000000a1`3f9ff6e0 00000000`00000000     : 00000000`00000032 00000000`00000000 00000000`00000000 000001eb`976973e0 : HID+0x24ad3
SYMBOL_NAME:  HID+24ad3
MODULE_NAME: HID
IMAGE_NAME:  HID.node
STACK_COMMAND:  ~37s; .ecxr ; kb
FAILURE_BUCKET_ID:  INVALID_POINTER_READ_c0000005_HID.node!Unknown
```

If there is any information that I need to add, please let me know as soon as possible, thanks!

## Dump files
[dumps.zip](https://github.com/node-hid/node-hid/files/8059663/dumps.zip)

todbot, 2022-02-14:
Generally weird crashes like this can happen if you've got two different threads trying to access the same HID device. 
If you've got native code calling into `node-hid`, I instead recommend you skip `node-hid` entirely and use `hidapi` directly.  `node-hid` is just a thin wrapper on `hidapi`.

Vict0r-Chen, 2022-02-15:
> Generally weird crashes like this can happen if you've got two different threads trying to access the same HID device. If you've got native code calling into `node-hid`, I instead recommend you skip `node-hid` entirely and use `hidapi` directly. `node-hid` is just a thin wrapper on `hidapi`.

Thank you for your reply!
For some reason I am currently not authorized to make changes to the C++ class library. When I attach the `hid.pdb` file to windbg, I see more detailed information:
```
STACK_TEXT:  
000000ea`7bfff9e0 00007ffc`88c57cc2     : 000002e0`00000040 000002e0`b0ef2a30 00000000`00000001 000000ea`7bfffa68 : HID!hid_read_timeout+0x103
000000ea`7bfffa40 00007ffc`88c5977e     : 00000000`00000000 00007ff7`99babd20 00007ff7`99babd20 00007ff7`99babcf8 : HID!ReadWorker::Execute+0x52
000000ea`7bfffaa0 00007ff7`93961cd2     : 00007ff7`99babd20 00002678`01660440 00007ff7`99babd20 00007ff7`99babcf8 : HID!Napi::AsyncWorker::OnExecute+0x1e
000000ea`7bfffb10 00007ff7`99babd20     : 00002678`01660440 00007ff7`99babd20 00007ff7`99babcf8 00007ff7`93961b60 : Upgrade_Test+0x1eb1cd2
000000ea`7bfffb18 00002678`01660440     : 00007ff7`99babd20 00007ff7`99babcf8 00007ff7`93961b60 00002678`01655b00 : Upgrade_Test+0x80fbd20
000000ea`7bfffb20 00007ff7`99babd20     : 00007ff7`99babcf8 00007ff7`93961b60 00002678`01655b00 00000000`00000000 : 0x00002678`01660440
000000ea`7bfffb28 00007ff7`99babcf8     : 00007ff7`93961b60 00002678`01655b00 00000000`00000000 00000000`00000b28 : Upgrade_Test+0x80fbd20
000000ea`7bfffb30 00007ff7`93961b60     : 00002678`01655b00 00000000`00000000 00000000`00000b28 000000ea`69ffce68 : Upgrade_Test+0x80fbcf8
000000ea`7bfffb38 00002678`01655b00     : 00000000`00000000 00000000`00000b28 000000ea`69ffce68 00000000`00000000 : Upgrade_Test+0x1eb1b60
000000ea`7bfffb40 00000000`00000000     : 00000000`00000b28 000000ea`69ffce68 00000000`00000000 00000000`00000000 : 0x00002678`01655b00

FAULTING_SOURCE_LINE:  E:\App\node_modules\node-hid\hidapi\windows\hid.c
FAULTING_SOURCE_FILE:  E:\App\node_modules\node-hid\hidapi\windows\hid.c
FAULTING_SOURCE_LINE_NUMBER:  776
SYMBOL_NAME:  HID!hid_read_timeout+103
MODULE_NAME: HID
IMAGE_NAME:  HID.node
STACK_COMMAND:  ~36s; .ecxr ; kb
FAILURE_BUCKET_ID:  INVALID_POINTER_READ_c0000005_HID.node!hid_read_timeout
OSPLATFORM_TYPE:  x64
```
The error seems to point to [this line](https://github.com/libusb/hidapi/blob/f6d0073fcddbdda24549199445e844971d3c9cef/windows/hid.c#L776) of code.
Is it possible to catch this error?

todbot, 2022-02-15:
That's in `hidapi`.  This is `node-hid`.  You should ask `hidapi` devs. 

But I suspect you won't get a good response because I think the reason this is happening is some other thread in your code is messing with the `node-hid` objects.

Vict0r-Chen, 2022-02-21:
> That's in `hidapi`. This is `node-hid`. You should ask `hidapi` devs.
> 
> But I suspect you won't get a good response because I think the reason this is happening is some other thread in your code is messing with the `node-hid` objects.

Thanks again for your reply.

I followed your suggestion to use `hidapi` directly to communicate with the device, and use `HID.close()` to close the handle in use before using `hidapi`, unfortunately the app still crashes.
But I replaced `HID.close()` with `HID.removeAllListeners()` and `HID.pause()`, then the problem is solved.

I didn't figure out why, but it did. In the past, the probability of crashing was 10%, and now I have tried 1000 times and it has not appeared once.

closed 2022-02-23 by Vict0r-Chen