node-hid/node-hid #463: HID Interface string descriptor product

adriencarbonaro, 2022-02-22:
I've seen a strange behavior with HID device enumeration with `node-hid`.

## Context

- I use `iProduct` string descriptor to set my product name.
- I use HID Interface `iInterface` to describe HID interface with a name.

## Current behavior

- When I set the `iInterface` string descriptor in HID Interface descriptor, `node-hid` use it as product key. But the actual `iProduct` string descriptor is also defined.
- Windows resolves the `iProduct` without issue and also resolves the HID `iInterface` string descriptor.

## Wanted behavior

- I think `node-hid` should resolve something like :
```
{
  "product": iProduct string descriptor,
  "hid": iInterface string descriptor,
}
```

## Descriptors

```
Device Descriptor:
------------------------------
0x12      bLength
0x01      bDescriptorType
0x0200    bcdUSB
0xEF      bDeviceClass
0x02      bDeviceSubClass   
0x01      bDeviceProtocol   
0x40      bMaxPacketSize0
0x0001    idVendor
0x0002    idProduct
0x0200    bcdDevice
0x01      iManufacturer       "Me"
0x02      iProduct            "ProductName"
0x03      iSerialNumber       "1234567890"
0x01      bNumConfigurations

[...]

Interface Descriptor:
------------------------------
0x09      bLength
0x04      bDescriptorType
0x03      bInterfaceNumber
0x00      bAlternateSetting
0x02      bNumEndPoints
0x03      bInterfaceClass     (Human Interface Device Class)
0x00      bInterfaceSubClass   
0x00      bInterfaceProtocol   
0x07      iInterface          "HIDName"

[...]

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Me"
0x02   0x0409  "ProductName"
0x03   0x0409  "1234567890"
0x06   0x0409  "Other Interface string desc"
0x07   0x0409  "HIDName"
```

todbot, 2022-02-22:
`node-hid` is just a thin wrapper on `hidapi`. I don't believe `hidapi` has access to USB interface names as HID works on both BLE and USB, and I'm unfamiliar with BLE and if it has the concept of interface name.

Typically in HID, if one is looking for a particular sub-interface of a HID device, that is keyed by the `usagePage` and `usage` parameters of the device's HID Report Descriptor.

If you feel you still need interface name, I recommend opening an issue in `libusb/hidapi`.

adriencarbonaro, 2022-02-22:
I would understand that iInterface is not accessible (even though I think it should as Windows is able to read at), but the fact that product key is swapped by iInterface seems like a bug to me. When I remove iInterface I get the product key as iProduct (as it should always be).

Are you telling me that hidapi is the origin of the bug ? In that case I will open an issue there.

todbot, 2022-02-22:
Yes, `node-hid` is just calling the `hidapi` function `hid_get_product_string()`, see: https://github.com/node-hid/node-hid/blob/master/src/HID.cc#L443

adriencarbonaro, 2022-02-23:
- I created an issue in `libusb/hidapi`.
- Would it be possible to expose the `hid_get_indexed_string` which also exists ?

todbot, 2022-02-23:
The API exists for `hid_get_indexed_string()` but the implementation only exists for Windows and libusb Linux (which is the lesser-used Linux impl, most people use the hidraw driver, and hidraw cannot get at that data).  I feel 
 adding it to `node-hid` would cause confusion.

mcuee, 2023-05-06:
Relevant hidapi issue has been closed.
* https://github.com/libusb/hidapi/issues/387

Therefore I think this isue can be closed as well.