node-hid/node-hid #467: Not able to install on Windows 10

anilsagar, 2022-04-21:
Getting the following error while installing from the palette:

`npm ERR! path C:\Users\anils\.node-red\node_modules\node-red-contrib-usbhid\node_modules\node-hid
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c prebuild-install || node-gyp rebuild
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@3.8.0
npm ERR! gyp info using node@14.16.0 | win32 | x64
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: Command failed: C:\Python39\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];
npm ERR! gyp ERR! stack   File "<string>", line 1
npm ERR! gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
npm ERR! gyp ERR! stack                       ^
npm ERR! gyp ERR! stack SyntaxError: invalid syntax
npm ERR! gyp ERR! stack
npm ERR! gyp ERR! stack     at ChildProcess.exithandler (child_process.js:308:12)
npm ERR! gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
npm ERR! gyp ERR! stack     at maybeClose (internal/child_process.js:1048:16)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
npm ERR! gyp ERR! System Windows_NT 10.0.19043
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\anils\\.node-red\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd C:\Users\anils\.node-red\node_modules\node-red-contrib-usbhid\node_modules\node-hid
npm ERR! gyp ERR! node -v v14.16.0
npm ERR! gyp ERR! node-gyp -v v3.8.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\anils\AppData\Local\npm-cache\_logs\2022-04-21T04_55_07_723Z-debug.log

21 Apr 10:25:07 - [warn] ------------------------------------------
Error: Install failed
    at C:\Users\anils\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\registry\lib\installer.js:285:25
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
21 Apr 10:25:07 - [error] Error: Install failed`

todbot, 2022-04-21:
Please provide:
- What version of `node-hid` you're trying to install
- What OS you're installing on
- The exact method you're using to attempt to install (what is "the palette")

anilsagar, 2022-04-21:
I am installing node-red-contrib-usbhid 0.0.1 from node-red "manage palette" (screenshot attached), on windows 10.
[image]

Julusian, 2022-04-21:
That node-red library looks to be using `"node-hid": "^0.5.1"`. So I'm not surprised that it doesn't work.

You might have some luck by following the steps to setup building node-hid from source https://github.com/node-hid/node-hid/tree/v0.5.7#compiling-from-source, or you could fork that node-red package and update node-hid to a modern version that will install much easier

anilsagar, 2022-04-21:
Got some success. Shared [here](https://discourse.nodered.org/t/accessing-usb-based-hid-rfid-reader-in-node-red/61597?u=anilsagar)

closed 2022-04-21 by anilsagar

todbot, 2022-04-21:
Who controls that list of libraries in that node-red "manage palette"?  Five years is a long time in NodeJs development.

todbot, 2022-04-21:
Also note that unless your USB HID RFID reader has a "hid data" mode and is instead acting like a keyboard (most act this way), then you will not be able to use it with `node-hid`.  The OS owns HID keyboards.