node-hid/node-hid #468: Error running install script

gawainb, 2022-05-02:
Getting this error when doing a `yarn install`
```
warning Error running install script for optional dependency: "/Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid-noevents/node_modules/node-hid, /Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid/node_modules/node-hid: Command failed.
Exit code: 1
Command: prebuild-install || node-gyp rebuild
Arguments: 
Directory: /Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid-noevents/node_modules/node-hid
Output:
prebuild-install WARN install No prebuilt binaries found (target=16.14.0 runtime=node arch=arm64 libc= platform=darwin)
gyp info it worked if it ends with ok
gyp info using node-gyp@9.0.0
gyp info using node@16.14.0 | darwin | arm64
gyp info find Python using Python version 3.10.4 found at \"/Library/Frameworks/Python.framework/Versions/3.10/bin/python3\"
gyp info spawn /Library/Frameworks/Python.framework/Versions/3.10/bin/python3
gyp info spawn args [
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid-noevents/node_modules/node-hid/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/gawainbracyii/Library/Caches/node-gyp/16.14.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/gawainbracyii/Library/Caches/node-gyp/16.14.0',
gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/Users/gawainbracyii/Library/Caches/node-gyp/16.14.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid-noevents/node_modules/node-hid',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc:612:31: error: too few arguments to function call, expected 3, have 2
  node::AtExit(deinitialize, 0);
  ~~~~~~~~~~~~                ^
/Users/gawainbracyii/Library/Caches/node-gyp/16.14.0/include/node/node.h:934:18: note: 'AtExit' declared here
NODE_EXTERN void AtExit(Environment* env,
                 ^
1 error generated.
make: *** [Release/obj.target/HID/src/HID.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:520:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
gyp ERR! System Darwin 21.4.0
gyp ERR! command \"/usr/local/bin/node\" \"/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js\" \"rebuild\"
gyp ERR! cwd /Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid-noevents/node_modules/node-hid
```

Julusian, 2022-05-02:
What version of node-hid is it installing? The file at `/Users/gawainbracyii/Developer/creative-ui-2.2/node_modules/@ledgerhq/hw-transport-node-hid-noevents/node_modules/node-hid/package.json` will tell you

Julusian, 2022-05-03:
Yeah I suspected it would be something like that.

That version of node-hid is over 3 years old and does is known to not work with recent versions of node. You will need to update the library that are pulling in that old version to resolve this

mcuee, 2023-05-04:
In this case, this issue can be closed.

closed 2023-11-25 by todbot