node-hid/node-hid #470: Publish linux-arm64 binaries

Julusian, 2022-05-18:
Any chance of prebuilds for linux-arm64 being added?

Im curious, how are you producing these prebuilds currently? Anything I could help automate via some github actions scripting?  Perhaps something from https://github.com/prebuild/prebuildify-cross could be borrowed for 'cross-compiling' linux binaries? Doing the rest in github actions is much simpler

todbot, 2022-09-28:
There's currently a `node-hid-v2.1.1-napi-v3-linux-arm.tar.gz` prebuild. Is that not correct?

todbot, 2022-09-28:
Oh no that may be Raspberry Pi.  Well I'll find out shortly.  Ignore this.

Julusian, 2022-09-28:
This is for the arm64 version of raspbian, not the same as the existing arm build.
Essentially the x86 vs x64 of the arm world.

In some other native libraries I am finding it really nice to produce all the builds in github actions, and it can use cpu emulation for arm/arm64 builds. It's been a few months and no complaints yet (with some users definitely using arm64 builds)

Julusian, 2023-01-08:
Closing as it looks like there is one for v2.1.2

closed 2023-01-08 by Julusian