node-hid/node-hid #472: No exclusive device access feature?

readandwrite864, 2022-05-25:
Say I have an RFID hid reader, and I want my app to exclusively possess this reader, so any other process doesn't receive symbols coming from that reader. I cannot see any API available to address this issue

Julusian, 2022-05-25:
This is a known limitation of the underlying hidapi library that this uses. Different OSes are opening the devices differently, and there is not yet a way to open as exclusive on windows. 
There was a pr recently for opening as shared on macos, so it looks likely that they would be willing to support opening exclusive on windows.
Further reading: https://github.com/libusb/hidapi/pull/397#issuecomment-1098007088 and the other linked issues

readandwrite864, 2022-05-25:
@Julusian Ok, so it seems to be a proper crossplatform support issue. I should have mentioned that my app is intended to work on linux.  Thank you for clarification.

I ended up using [node-exclusive-keyboard](https://github.com/pfirpfel/node-exclusive-keyboard), which does what i need via `EVIOCGRAB`

closed 2022-05-25 by readandwrite864

Julusian, 2022-05-25:
Im pretty sure that on linux node-hid is opening devices in exclusive mode, so it should work for you with that. 
But it sounds like the reader you are using acts like a keyboard so might not be compatible with node-hid anyway