node-hid/node-hid #477: With Electron, is it able to even if the window is not focused

Jannchie, 2022-06-30:


todbot, 2022-06-30:
Detecting HID input without the window being active is possible, yes.

I'm not sure what you mean by "it triggers an event". You mean your app or the device? The device does not know anything about window status.

Jannchie, 2022-06-30:
@todbot Thanks for the quick reply!

I use the following code to get the data:

``` js
for (const device of HID.devices()) {
  console.log(device.vendorId, device.productId, device.product);
  if (device.product === 'Controller (Xbox One For Windows)') {
    const hid = new HID.HID(device.vendorId, device.productId);
    hid.on('data', (data: Buffer) => {
      console.log(data.toString('hex'));
    });
  }
}
```

When my focus was on the window, it was able to get the data successfully. But when I move out of the window, I get a not-so-normal data (0080008000800080008000800000000000), and then no matter how I manipulate the gamepad, I can't get the data anymore.

todbot, 2022-06-30:
You will likely not be able to use an Xbox One controller with node-hid. That controller is not a normal HID device (I believe the Windows Xbox One driver creates a virtual HID device for apps). Also, since it's a gamepad, most OSes like Windows will "own" those devices and either not let your application work or work only partially.

You may have more luck with a browser-based gamepad API:
https://github.com/Tom32i/gamepad.js
https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API/Using_the_Gamepad_API

Jannchie, 2022-06-30:
Thank you very much for your reply!

closed 2022-06-30 by Jannchie