node-hid/node-hid #478: Cannot write to hid device

kuangping-kp, 2022-07-15:
var os = require('os');
var HID = require('node-hid');

var devices = HID.devices();
console.log(devices);

// var isStreamLightController = function (d) {
//   return d.vendorId === 0x1532 && d.productId === 0x0a24;
// };

// var isRawHidUsage = function (d) {
//   return d.usagePage === 0x0006 && d.usage === 0x0001;
// };

var deviceInfo = devices.find(function (d) {
  var isTeensy = d.vendorId === 0x5566 && d.productId === 0x0008;
  return isTeensy && d.usagePage === 0x0c && d.usage === 0x0001;
});
console.log(deviceInfo, 'ppppppppppppppppppppppppppp');
if (deviceInfo) {
  console.log(deviceInfo, '000000000000000000000000');

  var device = new HID.HID(0x5566, 0x0008);
  console.log(device);

  device.on('data', function (data) {
    console.log(data);
  });

  device.on('error', function (error) {
    console.log('22222222222222222222');
    console.log(error, 'eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee');
    device.close();
  });

  //   if (os.platform() == 'win32') {
  //     command.unshift(0x00);
  //   }

  // setTimeout(() => {
  let command = [
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00,
  ];
  console.log('Sending: ', command);
  device.write(command);
  // }, 200);

  device.close();
}

todbot, 2022-07-15:
Without knowing details of the device (e.g. what does it do, what is its HID report descriptor, what Usage Pages does it respond to? does it emulate a keyboard, mouse, or gamepad?), no assistance can be provided.

kuangping-kp, 2022-07-15:
模拟键盘      vid是 0x5566，pid是0x0008

kuangping-kp, 2022-07-15:
PC发送：0X55 + 0XFF + 0X00 + 0X00(数据长度)
MCU返回：0XAA + 0XFF + 0X00 + 0X01(数据长度) + 0X01/0X00(设置成功/设置失败)

kuangping-kp, 2022-07-15:
你好，我这个是想通过usb向键盘写入数据，并监听数据

Julusian, 2022-07-15:
It sounds like you might be trying to use a device that is not possible to use with node-hid due to platform limitations  https://github.com/node-hid/node-hid#devices-node-hid-cannot-read

kuangping-kp, 2022-07-15:
https://github.com/node-hid/node-hid#devices-node-hid-cannot-read  这里有解决方法？

kuangping-kp, 2022-07-15:
你好，可以详细的指出我的错误吗？

todbot, 2022-07-15:
You cannot read/write from HID keyboards with `node-hid`.  The OS controls the keyboard. This is for security reasons.  

If you are trying access a non-keyboard aspect of a keyboard, like built-in RGB lighting, that may show up as a separate `usagePage` and `usage` number in the device list. You must then filter on that, and only open up the device that has the exact vid, pid, usagePage, usage combination.

Even then you may run into trouble, if there is already a driver on the PC that is controlling those lights.

kuangping-kp, 2022-07-16:
你好，你这边说的出于安全原因是键盘没有给我权限？

kuangping-kp, 2022-07-16:
你好，你说的安全原因，是键盘没有给我权限？我的代码里面已经通过 vid、pid、usagePage、usage进行过滤了，就是我在调试的时候window系统是连接二个键盘的，就算连接一个也是写入失败的。

kuangping-kp, 2022-07-16:
你好，你说的安全原因，是键盘没有给我权限吗？

kuangping-kp, 2022-07-16:
我已经通过vid、pid、usagePage、usage 过滤了确切的设备，并将其打开了，但是就是一直写入失败。

kuangping-kp, 2022-07-16:
var isMyDevice = function (d) {
  return d.vendorId === 0x5566 && d.productId === 0x0008;
};

var isRawHidUsage = function (d) {
  return d.usagePage === 0x0c && d.usage === 0x0001;
};

var deviceInfo = devices.find(function (d) {
  return isMyDevice(d) && isRawHidUsage(d);
});

这是我的过滤代码

kuangping-kp, 2022-07-16:
你好，想问一下有没有可能是usb键盘的问题，导致数据不能写入

todbot, 2022-07-16:
HID UsagePage 0x0C is Consumer Control, the media keys on a keyboard. This usagepage is owned by the PC OS and not available to node-hid. 

You can see a list of usage pages here:
https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf

kuangping-kp, 2022-07-16:
HID UsagePage 0x0C 改成HID UsagePage 0x000C，也不可以吗？

kuangping-kp, 2022-07-16:
那我想和usb键盘通讯不用node-hid，还能用什么？

kuangping-kp, 2022-07-16:
是因为 HID UsagePage是0x0C，所以device.write才会写不进去吗，是不是HID UsagePage换一个值就可以啊

todbot, 2022-07-17:
> 那我想和usb键盘通讯不用node-hid，还能用什么？

I'm not sure how you can deal with media keys in node-hid. You can try using standard Javascript key reading APIs like the `window.addEventListener("keydown", function)` : https://developer.mozilla.org/en-US/docs/Web/API/Element/keydown_event

But media keys are trapped by the OS and not sent to the browser I think.

kuangping-kp, 2022-07-18:
window.addEventListener("keydown", function)这个不是监听按下时的状态吗，和媒体密钥有什么关系

kuangping-kp, 2022-07-19:
后台打印出来的_maxListeners: undefined

kuangping-kp, 2022-07-19:
写不进去和这个有原因？

kuangping-kp, 2022-07-22:
node-hid可以用来给键盘写入数据和读取数据吗？

todbot, 2022-07-22:
> node-hid可以用来给键盘写入数据和读取数据吗？

As mentioned previously many times: no, `node-hid` can cannot be used to read/write to keyboards.

closed 2022-07-22 by todbot