node-hid/node-hid #482: Missing serialNumber string

codemercs-com, 2022-07-25:
Hi,

I use the node-hid to communicate with multiple devices with the same VendorID and ProductID und use the serial number to get the path for the communication. 
I build my own node-red component for this on a raspberry pi 3+ (development machine) and it works as expected.

System:
_Linux 5.10.103-v7+ arm LE
Node-RED Version: v1.3.4
Node.js  Version: v10.24.0
Node-HID Version: 2.1.1
Libusb Version: 1.0.0.0 (inkl. dev)_

If I print out the device.find() object I get all informations about the device.

_{ vendorId: 1984,
  productId: 5377,
  path: '0001:0008:00',
  serialNumber: '000063EF',
  manufacturer: 'Code Mercenaries',
  product: 'IO-Warrior24',
  release: 4144,
  interface: 0 }
{ vendorId: 1984,
  productId: 5377,
  path: '0001:0008:01',
  serialNumber: '000063EF',
  manufacturer: 'Code Mercenaries',
  product: 'IO-Warrior24',
  release: 4144,
  interface: 1 }_

After I test this on a blank raspberry PI I missing most of the informations, most important the serial number.

System:
_Node-RED Version: v2.1.6
Node.js  Version: v12.22.12
Linux 5.15.32-v7+ arm LE
Node-HID Version: 2.1.1
Libusb Version: 1.0.0.0 (inkl. dev)_

  _{
    vendorId: 1984,
    productId: 5377,
    path: '0001:0004:00',
    release: 4144,
    interface: 0
  },
  {
    vendorId: 1984,
    productId: 5377,
    path: '0001:0004:01',
    release: 4144,
    interface: 1
  }_

Does anyone have any idea what is going wrong? Except for the versions of Node-Red and Node.js, the code is the same.
I try to install everything as a normal user and also as root. Same issues and same results. 

Thanks
Christoph

todbot, 2022-07-25:
It looks like you're using the `libusb` driverType for `node-hid` instead of the default `hidraw` driverType.
For some reason (this may be a limitation of `hidapi`, the library that `node-hid` wraps), Linux is not providing any information beyond VID/PID/path to hidapi when in libusb mode. 

You can try this out yourself for your Node project with `node-hid` installed by comparing the outputs of these two commands:
```
./node-modules/.bin/hid-showdevices libusb
./node-modules/.bin/hid-showdevices hidraw
```

This may be a bug in hidapi or `node-hid` because getting serialNumber should work. I'm looking into that now.

But as a workaround, can you do:  

`HID.setDriverType("hidraw")`

right after importing `node-hid`?

Julusian, 2022-07-25:
@todbot that was my initial thought, but then I noticed that both cases are using libusb. It is very easy to tell as a path of `0001:0008:00` is libusb, hidraw would be `/dev/hidraw1`.

If I try it locally on linux with libusb, I get the strings for some but not all devices. It looks to line up with whether a udev rule has been added to give my user permissions to the device.
@codemercs-com could you verify that you have matching udev rules on both setups? Or perhaps you are running as root on the dev pi but not the newer install?

codemercs-com, 2022-07-26:
@todbot I use libusb because hidraw show nothing:
`hid-showdevices hidraw
driverType: hidraw
devices: []`

The device has its own drivers and will not listed as "/dev/hidraw%". They have its own location "/dev/usb/iowarrior%" and the device drivers are part of the kernel sinde 2.6. I use the setting 'ID.setDriverType("libusb")'

After I install the node-hid global the output show the missing information (with sudo)

`sudo hid-showdevices libusb
driverType: libusb
devices: [
  {
    vendorId: 1984,
    productId: 5377,
    path: '0001:0004:00',
    serialNumber: '000062D0',
    manufacturer: 'Code Mercenaries',
    product: 'IO-Warrior24',
    release: 4144,
    interface: 0
  },
  {
    vendorId: 1984,
    productId: 5377,
    path: '0001:0004:01',
    serialNumber: '000062D0',
    manufacturer: 'Code Mercenaries',
    product: 'IO-Warrior24',
    release: 4144,
    interface: 1
  }
]`

But if I only run 'hid-showdevices libusb' I missing the same informations. So it looks like a permission conflict but I don't know why.

I have copy a rules file into /etc/udev/rules.d to get access for the 'pi' user without any root permissions.
`KERNEL=="iowarrior*", NAME="usb/iowarrior*", OWNER="pi", MODE="0666", GROUP="plugdev"`

I don't understand why it works on the older versions but not with the actual version of nodejs and node-red.

codemercs-com, 2022-07-26:
I found the issue!

After checking the permissions on my development machine in "dev/bus/usb/*" for the ports I saw that all the entries have the group plugdev and not root. 
After some digging I found a rules file which sets the "subsystems==usb" to this group "plugdev".  After I add this line into my rules file I have access to the other informations like serial number and product and my node will work. I will test this on an blank device and if all works

Complete rules file:
_KERNEL=="iowarrior*", NAME="usb/iowarrior*", OWNER="pi", MODE="0666", GROUP="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="07c0", ATTRS{idProduct}=="1501", GROUP="plugdev"_

Thanks for your help.

closed 2022-07-26 by codemercs-com