node-hid/node-hid #483: Receiving no data after pressing MUTE button on Jabra EVOLVE LINK and Yealink UH36

AleksadraMT, 2022-08-05:
Hello!

I'm having trouble reading data from the pressed MUTE button for Jabra EVOLVE LINK and Yealink UH36. Interesting that for Plantronics Blackwire 3220 Series everything works as expected.

I thought that I need to assign listeners for Phone Mute OOC button (checking this doc on the way: https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf), but it wasn't really helpful for now.

My devices usagePage is 12. 

Here is a small example of the demo code I implemented, trying to make it work.

`export class NodeHID {
  init = (): void => {
    console.log('NODE HID init');
    const devicesList = devices();

    console.log('DEVICES LIST', devicesList);

    // const hiddevice = devicesList.find(d => d.product?.toLocaleLowerCase().includes('plantronics'));
    // const hiddevice = devicesList.find(d => d.product?.toLocaleLowerCase().includes('yealink') && d.usagePage === 12);
    const hiddevice = devicesList.find(
      d => d.product?.toLocaleLowerCase().includes('jabra') && d.usagePage === 12
    );

    if (!hiddevice) return;

    console.log('CHOSEN DEVICE', hiddevice);
    console.log('deviceInfo.manufacturer:', hiddevice.manufacturer);
    console.log('deviceInfo.product:', hiddevice.product);
    console.log('deviceInfo.serialNumber:', hiddevice.serialNumber);

    const deviceHID = new HID(hiddevice.path!);

    deviceHID.on('data', function (data) {
      console.log('RECEIVED', data);
    });
  };
}`

Maybe node-hid is just not the good choice for what I'm trying to archive? Thank you for help.

todbot, 2022-08-05:
usagePage 12 (0x0C) is "Consumer Control", normally a part of keyboards and thus owned by the OS. 
`node-hid` has restricted to no ability to communicate with these devices since the OS controls them.

The technique usually employed with keyboards is to [use standard keyboard event listeners](https://developer.mozilla.org/en-US/docs/Web/API/Element/keydown_event).  With Consumer Control it's a bit more complicated, but I believe there are now a ["mute"/"unmute" events](https://developer.mozilla.org/en-US/docs/Web/API/MediaStreamTrack/mute_event) to hook into

AleksadraMT, 2022-08-09:
It's interesting... And what's about Telephony Device Page (0x0B)? Because in the device list I have the same devices under the usagePage 11 as well.

Thank you!

AleksadraMT, 2022-08-12:
Just made a few tests with Logitech H570e Stereo and for this one mute and answer the call are sending data, but not the volume controls (it's not necessary for now). Still interesting why mute/unmute work differently for different headsets, with the same usagePage 12 and usage 1.

I tried to use a hook for the MediaStreamTrack, but it doesn't work for my goals, because I'm not receiving mute/unmute events from Jabra and Yealink.

closed 2022-09-16 by AleksadraMT