node-hid/node-hid #485: error: architecture incompatibility

jackthta, 2022-09-22:
### Description
- I'm building on a project that utilizes the `node-hid` package, and I can't seem to figure out why I receive this error. It was suggested to me to try running [electron-hid-test](https://github.com/todbot/electron-hid-test) to see if it will work, but I received the same error (attached image). I'm assuming that if I can get [electron-hid-test](https://github.com/todbot/electron-hid-test) to work, it will on the project I'm working on as well.
- I'm running on a MacBook Pro M1 on macOS Monterey (12.4).

### Error
`mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))`

[image]

Would greatly appreciate any help!

Julusian, 2022-09-22:
What architecture is the version of node you have installed?

It looks like `prebuild-install` installed the x86_64 build for you, which you are then trying to load into arm64 electron and receiving this error. `prebuild-install` will install the version for whatever architecture of nodejs you ran the install with, so I think it likely that you have the x86_64 version installed

jackthta, 2022-09-22:
> What architecture is the version of node you have installed?
>
> It looks like `prebuild-install` installed the x86_64 build for you, which you are then trying to load into arm64 electron and receiving [...]

Apparently it's `x64`. Is it supposed to be `arm64` instead?

Julusian, 2022-09-23:
Yes, you should install the arm64 version

jackthta, 2022-09-27:
> Yes, you should install the arm64 version

Thank you so much, that fixed it!

closed 2022-09-27 by jackthta