node-hid/node-hid #487: Unhandled C++ Exception / 0x7ff8f8034fd9

hrueger, 2022-09-29:
Hi,
I just got this crash report in Sentry, and since hidapi has a function called `hid_get_feature_report`, the error probably comes from this lib?
https://sentry.internal.makepro-x.com/share/issue/aa0b31f57375418ea238679cb0f2ecb8/
My application is quite complex, that means I unfortunately don't have any code available to easily reproduce this...

Does the stack trace give you any hint what the problem could be?

todbot, 2022-09-29:
Hi @hrueger,

Unfortunately that stack trace isn't very helpful. The `hid_get_feature_report()` is actually from `hidapi`, of which `node-hid` is a thin wrapper on.  But the problem I would say lies in the Node realm, not hidapi's fault.

What version of `node-hid` is this app running?  The first step would be to get it at the most recent version.

I think I've seen this type of error when `node-hid` is used in a multi-threaded way and care isn't exercised when accessing a opened device in a separate thread. Since `hidapi` isn't thread-safe in general, `node-hid` doesn't attempt to be either. Please ensure all accesses to HID devices are in the same thread, and don't try to share devices across the Electron main/renderer process divide.

Julusian, 2022-09-29:
Yeah that stack trace looks very wrong, we are barely anything async, and definitely are not doing anything inside of `AfterThreadPoolWork`. Either that stack trace is badly mangled, or could that be something around when a worker thread is being destroyed?

hrueger, 2022-09-29:
Thanks for the fast responses.

> What version of node-hid is this app running?
I'm using version `2.1.0`.

I'm not using `node-hid` in a worker thread. It's only in the electron-main tread. I'm using it indirectly, the "parent" lib is `@elgato-stream-deck/node`.

The application itself uses a worker_thread, though, but I'm not loading `@elgato-stream-deck/node` nor `node-hid` there.