node-hid/node-hid #489: read duplicated result sometimes in vs code extension

xucs007, 2022-10-18:
hidapi/windows/hid.c  hid_read_timeout(), when used in VS Code extension, sometimes
get duplicated result from last command I sent to MCU. 
Most time results are correct, about 90 - 95% probably.

Julusian, 2022-10-18:
Are you making these `hid_read_timeout()` calls so that multiple are executing at once, or waiting for one to complete before doing the second?

xucs007, 2022-10-18:
Waiting for one to complete, before doing the seconds. Time interval between commands is around milliseconds of less.

xucs007, 2022-10-31:
I found another web forum, talking about same/similar issue: 

https://giter.vip/libusb/hidapi/issues/56

"For single write, it works fine. But when we write in a loop for about 100/500 times without any delay, it hangs in the write method."

"USB HID can only support one report every 1 millisecond." 

-- So it looks like some low-end/low-speed MCU, might not be able to handle USB-HID protocol well enough. Then what's the best solution?

todbot, 2022-10-31:
If you're writing so fast you're overwhelming the device causing it to hang the USB bus, there's not much node-hid can do about that.  That's a hardware problem.

xucs007, 2022-11-03:
My case may differ a little -- I got duplicated reading, not hang. 

I searched around, it seems usb offer 4 kinds of transfer protocol: transfer, controlTransfer, interruptTransfer, isochronousTransfer (https://developer.mozilla.org/en-US/docs/Web/API/USBDevice/). 
It seems to me it is due to my MCU detects error/timeout, then re-send, but node-hid can't tell error/timeout ?

xucs007, 2022-11-17:
We've found out the duplicated frame read was from a bug in our firmware, not from node-hid.  

Still, node.js connection behavior, such as waiting time (async/timeout), might not be very accurate or as expected. We didn't discover this duplicated frame problem when we connected via a C# desktop application.

closed 2022-11-17 by xucs007