node-hid/node-hid #493: MS Teams performance issues with Roccat Kone mouse

muellerj93, 2022-12-09:
Hi

Disclaimer:
I'm not sure whether my problem is a problem of node-hid or MS teams.
However, as I dug around the internet and googled for the file that's causing me trouble (after the forum post https://answers.microsoft.com/en-us/msteams/forum/all/ms-teams-helper-using-up-100-of-cpu-macbook/9305b96e-5381-43cf-88c4-d3ba39b37a58?page=21 hinted me at it), I found this github repo and thought better be safe than sorry and post about it, maybe there's an issue in this module.
So if you say this issue is not related to this module at all, the issue can be closed^^.

So long story short, I'm using MS Teams on my MacBook M1 Pro (company device) and noticed, that whenever I'm working from home (with my own mouse / keyboard) that teams' performance is so terrible, it's unusable.

I figured out by trial and error and unplugging / plugging devices back in, that it was either my mouse or keyboard (which is strange enough) that causes a "ms teams helper (renderer)" process cpu load to skyrocket.

As mentioned above, after digging around in the internet, I found the forum post:

https://answers.microsoft.com/en-us/msteams/forum/all/ms-teams-helper-using-up-100-of-cpu-macbook/9305b96e-5381-43cf-88c4-d3ba39b37a58?page=21

which pointed out a possible solution, which was to rename a certain file:

`sudo mv "/Applications/Microsoft Teams.app/Contents/Resources/app.asar.unpacked/node_modules/media-hid/build/Release/media-hid.node" "/Applications/Microsoft Teams.app/Contents/Resources/app.asar.unpacked/node_modules/media-hid/build/Release/media-hid.node.bak"`

Sure, this is not really a good or safe workaround to disable teams from loading a file it depends on, however teams is unusable without this change for me.

So after googling for the file name, I ended up here posting this, in case this file belongs to your module.

For reference, the mouse I'm using is a https://www.amazon.de/ROCCAT-Kone-Pure-Performance-schwarz/dp/B009DYOPJQ

Looking forward to your feedback (positive or negative)
BR

Julusian, 2022-12-09:
Based on the path of the file you moved, that is referring to a different npm library called `media-hid`.

Looking that up on npm gives an interesting result.. https://www.npmjs.com/package/media-hid?activeTab=readme

muellerj93, 2022-12-09:
Ok thanks for the clarification.

How did you find the correct package? I can access it through the link you posted, but can't find it even if I search directly on npmjs.com.

Also I couldn't find any package information regarding app.asar, where did you get that from?

I'm just curious how I can analyse such things on my own.^^

Thanks in advance

Julusian, 2022-12-09:
> How did you find the correct package? I can access it through the link you posted, but can't find it even if I search directly on npmjs.com.

Yeah it didn't show up in the search for me, but I went to another package page and edited the url to look at that.

> Also I couldn't find any package information regarding app.asar, where did you get that from?

The `app.asar` file is part of your teams installation, located at `/Applications/Microsoft Teams.app/Contents/Resources/app.asar`. Think of it as a zip file (with different encoding) containing the code for the application.

I extracted the one on my machine with `npx asar e "/Applications/Microsoft Teams.app/Contents/Resources/app.asar" ~/teams-asar` (using the npm package `asar` to extract the asar file) extracting to a folder called teams-asar inside your home folder) and then looking at the files inside `~/teams-asar/node_modules/media-hid/`.

muellerj93, 2022-12-09:
Thanks :)
I'll close the issue then.

closed 2022-12-09 by muellerj93