node-hid/node-hid #494: Crash when the V8 Sandbox is enabled in Electron 21&22

oalfroukh, 2023-01-08:
Hi,
I have a crash in real-time when using `node-hide v2.1.2` in Electron `v21 & 22` while the same code is OK when using Electron `v19`, here is the Error I got:

```C
[14388:0108/141933.609:ERROR:node_bindings.cc(146)] Fatal error in V8: v8_ArrayBuffer_NewBackingStore When the V8 Sandbox is enabled, ArrayBuffer backing stores must be allocated inside 
the sandbox address space. Please use an appropriate ArrayBuffer::Allocator to allocate these buffers.
```

FYI, I'm using the library from the main process and the process is sandboxed.

Julusian, 2023-01-08:
Are you sure you are running v2.1.2? This was fixed in that release, and I don't see any other buffer allocations that could have been missed.

If you definitely are, could you try forcing node-hid to rebuild, in case that helps

oalfroukh, 2023-01-08:
Hi @Julusian 
Thanks for your response! I'm very sorry, I was using `v2.1.1`, I just checked `v2.1.2` the issue is fixed. I'll close the issue.

closed 2023-01-08 by oalfroukh