node-hid/node-hid #496: TypeError: could not send feature report to device

seflm, 2023-02-11:
Hello,

I have a custom keyboard based on AtMega32u4 MCU running a modified Arduino core. 

Using node-hid I am trying to send a feature report to the keyboard. I came across the following error:

```
Error occurred in handler for 'saveConfig': TypeError: could not send feature report to device
    at Device.sendConfig (C:\█████████\dist\backend\device.js:74:19)
    at DeviceManager.saveConfig (C:\█████████\dist\backend\DeviceManager.js:70:16)
    at C:\█████████\dist\backend\main.js:67:105
    at node:electron/js2c/browser_init:2:97574
    at EventEmitter.<anonymous> (node:electron/js2c/browser_init:2:81087)
    at EventEmitter.emit (node:events:513:28)
```

The thing is, the data are actually received by the device despite this error. 
And when trying to send the same data using the Device Monitoring Studio software, there is no error.

All this leads me to suspect there is a bug inside the node-hid. However, I might be missing something. Is node-hid may be waiting for some confirmation from the device? Afaik, this happens on a lower level. 

Note: the Arduino core's HID communication is a mess but I don't believe it is related.

<details>
<summary><b>Report descriptor here</b></summary>
<br>

```

    //  Keyboard
    0x05, 0x01,                      /* USAGE_PAGE (Generic Desktop)	  47 */
    0x09, 0x06,                      /* USAGE (Keyboard) */
    0xa1, 0x01,                      /* COLLECTION (Application) */
		0x85, 0x02,	 					 /*   REPORT_ID   */
		0x05, 0x07,                      /*   USAGE_PAGE (Keyboard) */

		// Keyboard Modifiers (shift, alt, ...)
		0x19, 0xe0,                      /*   USAGE_MINIMUM (Keyboard LeftControl) */
		0x29, 0xe7,                      /*   USAGE_MAXIMUM (Keyboard Right GUI) */
		0x15, 0x00,                      /*   LOGICAL_MINIMUM (0) */
		0x25, 0x01,                      /*   LOGICAL_MAXIMUM (1) */
		0x75, 0x01,                      /*   REPORT_SIZE (1) */
		0x95, 0x08,                      /*   REPORT_COUNT (8) */
		0x81, 0x02,                      /*   INPUT (Data,Var,Abs) */

		// Reserved byte
		0x95, 0x01,                    //   REPORT_COUNT (1)
		0x75, 0x08,                    //   REPORT_SIZE (8)
		0x81, 0x03,                    //   INPUT (Cnst,Var,Abs)

		// 6 keys at once on keyboard
		0x95, 0x06,                    //   REPORT_COUNT (6)
		0x75, 0x08,                    //   REPORT_SIZE (8)
		0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
		0x25, 0x73,                    //   LOGICAL_MAXIMUM (115)
		0x05, 0x07,                    //   USAGE_PAGE (Keyboard)
		0x19, 0x00,                    //   USAGE_MINIMUM (Reserved (no event indicated))
		0x29, 0x73,                    //   USAGE_MAXIMUM (Keyboard Application)
		0x81, 0x00,                    //   INPUT (Data,Ary,Abs)
    0xc0,                           // END_COLLECTION

	// Vendor defined usage
	0x06, 0x00, 0xff,              //   USAGE_PAGE (Vendor Defined Page 1)
    0x09, 0x01,                    //   USAGE (Vendor Usage 1)
	0xa1, 0x01,                    // COLLECTION (Application)

		//Custom Deivce Info Report
		0x85, 0x03,                    //   REPORT_ID (3)
		0x09, 0x01,                    //   USAGE (Vendor Usage 1)
    	0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    	0x26, 0xff, 0x00,			   //   LOGICAL_MAXIMUM (255)
    	0x75, 0x50,                    //   REPORT_SIZE (80)
    	0x95, 0x01,                    //   REPORT_COUNT (1)
    	0xB1, 0x01,                    //   FEATURE (Cnst,Ary,Abs)*/

        //Test Report
		0x85, 0x04,	 				   //   REPORT_ID  (4)
		0x09, 0x01,                    //   USAGE (Vendor Usage 1)
		0x95, 0x01,//0x16,                    //   REPORT_COUNT (22)
		0x75, 0x30,                    //   REPORT_SIZE (48)
		0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
		0x26, 0xff, 0x00,              //   LOGICAL_MAXIMUM (255)
		0xB1, 0x00,                    //   FEATURE (Data,Ary,Abs)*/
    0xc0                           // 	END_COLLECTION

```

</details>

The error occurs when trying to send report ID 4 using `this.conn.sendFeatureReport([0x04,0x00,0x01,0x01,0x01,0x01,0x01,0x01]);`

todbot, 2023-02-12:
This is not a problem with `node-hid`, but rather a quirk of Windows. 

On Windows, your report buffer must be exactly the same size as the report. For the HID Report Descriptor you list, with report 4, Windows is expecting a 1057 byte buffer: 1 for reportId  + 1056 byte report (22 fields of 48 bits each). If you don't give Windows exactly the size it wants, you will get an error. Different versions of Windows will choose to send or not send the report if it's not of the correct size.   There's no way for `node-hid` (or `hidapi`, the underlying library `node-hid` thinly wraps) to know the size of these reports definitively.  It's frustrating.

closed 2023-02-12 by todbot

seflm, 2023-02-12:
I don't believe this is the case for various reasons. 

- In the report descriptor, the REPORT_COUNT of 22 is commented out and replaced with 0x01 for testing purposes. The expected buffer length should be 6B or 7B with report ID. That should match the array being sent.
- The data are actually received by the device despite the error.
- Other software on the same Windows machine is able to send the same data without any error.
- Using a simple while I tried to `sendFeatureReport` with a variable length between 0 and 2000, and none of them passed without this error.

todbot, 2023-02-12:
You may be running into problems because the HID Report Descriptor specifies that each (now single) 48-bit field value can only range from 0 to 255 (LOGICAL_MINIMUM & LOGICA_MAXIMUM). 

That is, most all your values must be zero and only one byte can range from 0-255 (I believe the first byte).  When you send `0x01,0x01,0x01,0x01,0x01,0x01` you're actually telling Windows you want to send the value 1103823438081 in your 48-bit number.  Windows is very strict with HID Reports and is likely enforcing this.  

As a test, do you get the same error if you send this:
```js
this.conn.sendFeatureReport([0x04,0x00,0x00,0x00,0x00,0x00,0x00);
```

seflm, 2023-02-12:
Thank you for pointing this out. I understood this part of HID spec differently. 

However, it hasn't solved the issue. 
Calling `sendFeatureReport([0x04,0x00,0x00,0x00,0x00,0x00,0x00])` produces the error as well.

For now, I am ignoring the error as it otherwise works ok, and the device gets the data.

todbot, 2023-02-13:
Another thing to look into: are you opening the correct device?  In HID, each usage shows up as a different device.  Your HID Report Descriptor describes two usages: `usagePage=0x01; usage=0x06` (keyboard) and `usagePage=0xff00; usage=0x01`.  You will need to filter on this in addition to VID/PID, as they show up as distinct devices to `node-hid` and you are not guaranteed which usage you will get.  

The standard pattern to filter on the (vid,pid,usagePage,usage) looks like: 
```js
var devices = HID.devices();
var deviceInfo = devices.find( function(d) {
    var isMyDevice = d.vendorId===0x16C0 && d.productId===0x0486; // whatever your VID/PID are
    return isMyDevice && d.usagePage===0xFF00 && d.usage===0x0001;
});
if( deviceInfo ) {
  var device = new HID.HID( deviceInfo.path );
  // ... use device
}
```

seflm, 2023-02-13:
I rechecked and can confirm I am sending the feature report to the correct device (usagePage 0xFF00).