node-hid/node-hid #497: build on MacBook M2

TomFuch, 2023-02-15:
How can I build it on Darwin with M2 Macbook?

Julusian, 2023-02-15:
If you are using a recent version of node-hid (v2.1.1 or later. December 2020 or newer) then prebuilds are provided.
If you are using older than that, then it most likely will not be possible to compile it without making some changes to the library.

There is some documentation about compiling in the readme https://github.com/node-hid/node-hid#compiling-from-source

TomFuch, 2023-02-16:
Hi!

I have installed the latest Version. After a few tests I found out, that it is a problem if the path to my local repo contains whitespaces. I moved the repo to another path without spaces and all works imeadetly.

Thanks!

Julusian, 2023-02-28:
I've never tried making a universal build so I am not sure.

That is something I should test for https://github.com/node-hid/node-hid/pull/490 though. That might behave better or might not

oalfroukh, 2023-02-28:
seems Adding `'!node_modules/**/*.{mk,a,o,h}'`  to files in electron-builder config fixes the issue!
Thanks a lot for your response!

mcuee, 2023-05-04:
In this case, the issue can be closed.

closed 2023-11-25 by todbot