node-hid/node-hid #498: "Error in native callback"

joeatmevo, 2023-03-07:
Hi all, hoping to get some guidance here. I am working on a Streamdeck plugin for a light and confirmed my logic using Python just using the hid package there but since I can't use Python for Streamdeck SDK natively I am trying to use Node but am running into this native callback error. To reiterate, I was able to control this just fine with Python, so the product definitely wasn't fighting it. 

**System:** Windows 10
**Node Version:** 19.7.0
**Node-Hid Version:** 2.1.2

Ensured I used Admin Command Prompt to have admin rights to run it. Still nothing.

Hopefully I am just dumb and missing something but any guidance would be great.

`import hid

DEVICE_PRODUCT_NAME = "Code"
MESSAGE_HEX = '11ff041c000000000000000000000000000000000000000000000000000000'

# Find all connected HID devices
devices = hid.enumerate()

# Find the HID device with the specified product name
device_info = None
for device in devices:
    if DEVICE_PRODUCT_NAME in device['product_string']:
        device_info = device
        break

if not device_info:
    print('Device not found')
    exit(1)

# Connect to the HID device and open the HID interface
device_path = device_info['path']
device = hid.device()
device.open_path(device_path)
device.set_nonblocking(1)

# Send a message to the HID device
message = bytes.fromhex(MESSAGE_HEX)
device.write(message)

# Wait for a response from the HID device
response = b''
while not response:
    response = device.read(64)

if response:
    print(f"Received response: {bytes(response).hex()}")

# Close the connection to the HID device
device.close()
`

and it continues to throw this error.

`C:\LControl\node_modules\node-hid\nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: Error in native callback
    at new HID ←[90m(C:\LControl\←[39mnode_modules\←[4mnode-hid←[24m\nodehid.js:49:17←[90m)←[39m
    at Object.<anonymous> ←[90m(C:\LControl\←[39mapp.js:23:16←[90m)←[39m
←[90m    at Module._compile (node:internal/modules/cjs/loader:1275:14)←[39m
←[90m    at Module._extensions..js (node:internal/modules/cjs/loader:1329:10)←[39m
←[90m    at Module.load (node:internal/modules/cjs/loader:1133:32)←[39m
←[90m    at Module._load (node:internal/modules/cjs/loader:972:12)←[39m
←[90m    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:83:12)←[39m
←[90m    at node:internal/main/run_main_module:23:47←[39m`

todbot, 2023-03-07:
You should check out some of the "test-*" examples in the "src" directory to see how to properly create a node-hid object and find an device.  

Also, you may want to check out https://github.com/julusian/node-elgato-stream-deck to perhaps save you some work.

Jcapehart2, 2023-03-07:
> y

I'll do that, thank you! I apologize if I just completely missed the mark here. My goal definitely isn't to waste anyone's time! 

The node elgato stream deck project is confusingly named but is not related to streamdeck at all. Although some of the concepts may be the same it appears the goal of that project was to create a web based version of a streamdeck vs the hardware version.

todbot, 2023-03-07:
I am confused. Is the device you're trying to communicate with not the Elegato Stream Deck?

Jcapehart2, 2023-03-07:
> I am confused. Is the device you're trying to communicate with not the Elegato Stream Deck?

Sorry for the confusion, no, that's not what I am trying to communicate with. The streamdeck is a piece of hardware that you press a button and an action occurs. I am building a plugin for the streamdeck but I can't move forward with that part of the project until I identify where I am wrong on the node-hid package. 

That package also isn't related to the Elgato streamdeck in anyway. They used a poor name I can only assume for SEO purposes, but it's to create a digital alternative to that piece of hardware as stated in their Readme.

todbot, 2023-03-07:
Can you give a link describing this device so I can read up on it?

Jcapehart2, 2023-03-07:
Unfortunately not, as I'm testing this product out and it's not yet in the public. I just wanted to get ahead of the curve to offer the ability to control it when it is released. I do know though that I was able to successfully control it using Pythons HID package without issue so I don't know that it's really anything special with the method that it is being controlled. 

Does the way I am sending the message seem off to you for this package specifically? For all intents and purposes that message should successfully work based on the way this package is used from the way I see from the test scripts provided the package works with this device like the Python one does?If it comes down to the package just not playing nice, that's something I can live with I just need to identify if it's a me or a package/device issue. :P

todbot, 2023-03-07:
And again, please crib from the examples to show how to open HID devices.  There is no `'w+'` argument for instance to to `new HID.HID()`.  

As for determining which device to open (your original question), generally you first search by `vendorId` and `productId`, then filter by `usagePage` and `usage` to get the particular usage in the HID Report Descriptor of your device. (Most HID devices have multiple HID Reports they respond to and you must send on the correct one, particularly on Window, or you get weird errors from the OS)  An example of doing this is https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js

Jcapehart2, 2023-03-07:
Thank you for that, I'll keep reading into it and get more familiar. Thanks for the guidance and your time!

Jcapehart2, 2023-03-07:
Thank you, the test js did help. All is well.

closed 2023-03-07 by Jcapehart2

todbot, 2023-03-07:
Awesome!  Can't wait to learn about this new streamdeck device!