node-hid/node-hid #500: build error on Ubuntu 22.04

atillabirer, 2023-03-18:
`
      |   ^~~~~
../src/HID.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE HID::read(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/HID.cc:262:52: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  262 |   uv_queue_work(uv_default_loop(), req, recvAsync, (uv_after_work_cb)recvAsyncDone);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/HID.cc: In static member function ‘static void HID::Initialize(v8::Local<v8::Object>)’:
../src/HID.cc:612:16: error: cannot convert ‘void (*)(void*)’ to ‘node::Environment*’
  612 |   node::AtExit(deinitialize, 0);
      |                ^~~~~~~~~~~~
      |                |
      |                void (*)(void*)
In file included from ../src/HID.cc:34:
/home/atilla/.cache/node-gyp/16.19.1/include/node/node.h:979:38: note:   initializing argument 1 of ‘void node::AtExit(node::Environment*, void (*)(void*), void*)’
  979 | NODE_EXTERN void AtExit(Environment* env,
      |                         ~~~~~~~~~~~~~^~~
../src/HID.cc: At global scope:
/home/atilla/.cache/node-gyp/16.19.1/include/node/node.h:887:7: warning: cast between incompatible function types from ‘void (*)(v8::Local<v8::Object>)’ to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
  887 |       (node::addon_register_func) (regfunc),                          \\\n      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/atilla/.cache/node-gyp/16.19.1/include/node/node.h:921:3: note: in expansion of macro ‘NODE_MODULE_X’
  921 |   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
      |   ^~~~~~~~~~~~~
../src/HID.cc:646:3: note: in expansion of macro ‘NODE_MODULE’
  646 |   NODE_MODULE(HID, init);
      |   ^~~~~~~~~~~
make: *** [HID-hidraw.target.mk:124: Release/obj.target/HID-hidraw/src/HID.o] Error 1
make: Leaving directory '/home/atilla/stellaswap-interface/node_modules/node-hid/build'
`
OS: Ubuntu 22.04, libusb-dev and libhidapi-dev are installed
Kernel: 5.19.0-35

Julusian, 2023-03-18:
That error looks like it for v1 of node-hid, and does not line up with v2.
v2 was released over 2 years ago, and is api compatible with v1, so it should be easy to update and everything will just work

atillabirer, 2023-03-18:
replaced all v1 package references with v2, now compiles, thank you chef

mcuee, 2023-05-04:
So this can be closed, right?

mcuee, 2023-05-06:
BTW, I just tried to build node-hid under the older Ubuntu 20.04 version. The built-in nodejs version is too old. So I just download version 18.16.0 and then there is no problem to build node-hid.