node-hid/node-hid #502: Build error on Mac M1

Jater0, 2023-04-12:
hello guys, I wanted to clone the repo to develop it, there was a issue when I built it,   the script failed during run `npm run gypbuild` or `npm install --build-from-source`

> node: 18.15.0
> python 3.7
> node-gyp 6.1.0

logs shows
``` shell
> node-hid@2.1.1 gypbuild
> node-gyp build

gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@18.15.0 | darwin | x64
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
  LIBTOOL-STATIC Release/hidapi.a
env: python: No such file or directory
make: *** [Release/hidapi.a] Error 127
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (xxxx/node-hid/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:291:12)
gyp ERR! System Darwin 22.3.0
gyp ERR! command "/usr/local/bin/node" "xxxx/node-hid/node_modules/.bin/node-gyp" "build"
gyp ERR! cwd xxxx/node-hid
gyp ERR! node -v v18.15.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 
```

Julusian, 2023-04-12:
It looks like your clone is missing the submodule of hidapi.
when cloning did you specify the `--recursive` flag? You can ensure they are cloned with `git submodule init` and `git submodule update`

Jater0, 2023-04-13:
> It looks like your clone is missing the submodule of hidapi. when cloning did you specify the `--recursive` flag? You can ensure they are cloned with `git submodule init` and `git submodule update`

i was cloned without `--recursive` & i have hidapi submodule

Julusian, 2023-04-17:
Oh sorry, I misread the error log. 
`env: python: No such file or directory` is the relevant line.

How is your version of `node-gyp` installed? I expect that updating that to latest will resolve this.
I don't use macos, but I remember this being a reasonably common issue as macos changed something about python within the last couple of versions

cbeeler, 2023-05-04:
I am receiving this error as well, Ventura 13.3.1 and XCode 14.3. I'm assuming its because [Apple Removed Python 2.7.](https://developer.apple.com/documentation/macos-release-notes/macos-12_3-release-notes#Python)

`python3` exists on the `env` but not `python`. I'm assuming there is some way to direct `python` to `python3` in the build system.

Julusian, 2023-05-04:
Try following the instructions at https://github.com/nodejs/node-gyp/blob/main/docs/Updating-npm-bundled-node-gyp.md to install a newer node-gyp that should work for you.

I do question why something is deciding that node-hid needs recompiling, but that is a different matter

cbeeler, 2023-05-04:
> Try following the instructions at https://github.com/nodejs/node-gyp/blob/main/docs/Updating-npm-bundled-node-gyp.md to install a newer node-gyp that should work for you.

That worked for me! Thank you! I'm glad you knew that was an option. My guess is that because we are using an older node and the default npm that comes with it is not updated for newer macOS issues.

I'm thinking we drop the link you gave [in the readme here](https://github.com/node-hid/node-hid#mac-os-x-108) and call it a day. I don't think its your responsibility to modify global tools.

mcuee, 2023-05-04:
So this issue can be closed, right?

closed 2023-05-06 by Jater0