node-hid/node-hid #510: Windows build issue

mcuee, 2023-05-06:
Using a new installation of nodejs 18.16.0 under Windows 11 with Python 3.11 and VS2019 build tools, there are issues to build node-hid.

```
PS C:\work\libusb\node-hid> npm install --build-from-source

> node-hid@2.1.2 install
> prebuild-install --runtime napi || node-gyp rebuild

gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@18.16.0 | win32 | x64
gyp info find Python using Python version 3.11.3 found at "C:\Users\xiaof\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\python.exe"
gyp info find VS using VS2019 (16.11.33529.622) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools"
gyp info find VS run with --verbose for detailed information
(node:9616) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
(Use `node --trace-deprecation ...` to show where the warning was created)
gyp info spawn C:\Users\xiaof\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\python.exe
gyp info spawn args [
gyp info spawn args   'C:\\work\\libusb\\node-hid\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\work\\libusb\\node-hid\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\work\\libusb\\node-hid\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\xiaof\\AppData\\Local\\node-gyp\\Cache\\18.16.0\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\xiaof\\AppData\\Local\\node-gyp\\Cache\\18.16.0',
gyp info spawn args   '-Dnode_gyp_dir=C:\\work\\libusb\\node-hid\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\xiaof\\\\AppData\\\\Local\\\\node-gyp\\\\Cache\\\\18.16.0\\\\<(target_arch)\\\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\work\\libusb\\node-hid',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\work\\libusb\\node-hid\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
Traceback (most recent call last):
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\gyp_main.py", line 50, in <module>
    sys.exit(gyp.script_main())
             ^^^^^^^^^^^^^^^^^
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 554, in script_main
    return main(sys.argv[1:])
           ^^^^^^^^^^^^^^^^^^
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 547, in main
    return gyp_main(args)
           ^^^^^^^^^^^^^^
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 520, in gyp_main
    [generator, flat_list, targets, data] = Load(
                                            ^^^^^
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 136, in Load
    result = gyp.input.Load(build_files, default_variables, includes[:],
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\input.py", line 2782, in Load
    LoadTargetBuildFile(build_file, data, aux_data,
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\input.py", line 391, in LoadTargetBuildFile
    build_file_data = LoadOneBuildFile(build_file_path, data, aux_data,
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\input.py", line 234, in LoadOneBuildFile
    build_file_contents = open(build_file_path, 'rU').read()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid mode: 'rU' while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\work\libusb\node-hid\node_modules\node-gyp\lib\configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:291:12)
gyp ERR! System Windows_NT 10.0.22621
gyp ERR! command "C:\\nodejs\\node.exe" "C:\\work\\libusb\\node-hid\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\work\libusb\node-hid
gyp ERR! node -v v18.16.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok
npm ERR! code 1
npm ERR! path C:\work\libusb\node-hid
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c prebuild-install --runtime napi || node-gyp rebuild

npm ERR! A complete log of this run can be found in: C:\Users\xiaof\AppData\Local\npm-cache\_logs\2023-05-06T01_20_46_047Z-debug-0.log
```

mcuee, 2023-05-06:
Google shows that it is a known issue for node-gyp.
* https://github.com/nodejs/node-gyp/issues/2219

Workaround is to change `rU` to `r` as per the above build log.
File "C:\work\libusb\node-hid\node_modules\node-gyp\gyp\pylib\gyp\input.py", line 234

todbot, 2023-11-25:
Sounds like not a node-hid issue. I updated the `prebuild` package which I think updates `node-gyp` and now I cannot replicate this issue on Windows 11 or in CI.

closed 2023-11-25 by todbot