node-hid/node-hid #511: Troubleshooting Electron and node-hid: Unable to Read from USB RFID Reader in the App (the code works)

echtme, 2023-05-16:
I'm using node-hid to read data from a USB RFID card reader.

The code runs perfectly when I start with "npm run start." However, when I create the Electron App and run it, I get an error (see below). I'm not sure if this is caused by accessing the USB, which requires sudo on a Mac.

Note: By default, the RFID reader is listed as a keyboard. Scanning a card prints out the ID as keyboard output.

**MacOSX**: 13.3.1
**Electron**: 18.0.1
**node-hid**: 2.1.1

Reader Snipped:
```javascript
const HID = require("node-hid"); 
const VENDOR_ID = 65535;`
const PRODUCT_ID = 53;
const device = new HID.HID(VENDOR_ID, PRODUCT_ID);

device.on("data", function (data) {
  console.log(data);
});

device.on("error", function (err) {
  console.log(err);
});
``` 

Package.json
```javascript
"scripts": {
    ...
    "start": "sudo electron .",
    ...
  },
``` 

  
  
  Electron App start error:
  `Uncaught Exception:
TypeError: cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS05@14100000/USB2.0 HUB@14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14130000/USB Reader@14130000/Keyboard@0/AppleUserUSBHostHIDDevice
at new HID (/Users/Tester/Development/Dev-Tests/dist/mac/RFID.app/Contents/Resources/app.asar/node_modules/node-hid/nodehid.js:49:17)
at /Users/Tester/Development/Dev-Tests/dist/mac/RFID.app/Contents/Resources/app.asar/serverUSB_Reader.js:43:18
at Array.forEach (<anonymous>)
at Object.<anonymous> (/Users/Tester/Development/Dev-Tests/dist/mac/RFID.app/Contents/Resources/app.asar/serverUSB_Reader.js:42:13)
at Module._compile (node:internal/modules/cjs/loader:1116:14)
at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10)
at Module.load (node:internal/modules/cjs/loader:988:32)
at Module._load (node:internal/modules/cjs/loader:829:12)
at Function.c._load (node:electron/js2c/asar_bundle:5:13343)
at Module.require (node:internal/modules/cjs/loader:1012:19)`

JunzheFan, 2023-05-17:
perhaps you need to elevate the electron app priviliage ? https://stackoverflow.com/questions/37322862/check-if-electron-app-is-launched-with-admin-privileges-on-windows ,  if you are using electron, why not just use the webhid offered by broswer ?

echtme, 2023-05-17:
I looked into WebHID, and I can successfully connect to the device. However, I'm facing difficulties in retrieving data from the reader. 

Currently, I can obtain the scanned card data in vanilla JS by using an event listener for 'keydown.' However, I need to identify the specific device from which the data is coming. I have three of these readers below, and it is crucial for me to recognize which one triggered the event. Obtaining the card ID is not as important yet.

Is there way to get the triggered device ID (1, 2 or 3)?

My RFID reader: https://www.amazon.com/Reader-125KHz-Proximity-Keyboard-Android/dp/B07TMNZPXK/ref=sr_1_5?crid=2NM7CW1B8GS98&keywords=rfid+card+reader&qid=1684306221&sprefix=rfid+card+reade%2Caps%2C277&sr=8-5

JunzheFan, 2023-05-17:
As far as I understand, you have three identical devices/readers connected to your pc, and you want to know which one sends out the data, in this case I think you need to make an internal identifier in code like uuid to track, unless you can configure your reader to send the device id along with each input report

echtme, 2023-05-17:
Hey, thanks for the quick response.

Yes, in a nutshell, this is what I want to do. With node-hid, I'm able to identify the devices by ID. However, node-hid requires sudo, and this doesn't when compiling the Electron app for me.

Setup:
(HTML receiving data) <-------- (express + socket) <-------- (RFID reader node-hid)

I can compile and build an Electron with the HTML and express + socket, and run the reader script separately. That works, but I need a standalone app for this.

todbot, 2023-05-17:
In general  you [cannot read HID devices that act as keyboards](https://github.com/node-hid/node-hid#devices-node-hid-cannot-read). When you're running as root, you're subverting the OS's input system security measures. 

Some USB HID RFID readers have a "USB HID data" mode (or "USB Serial" mode), and when in those modes, you could then use `node-hid` (or `node-serialport`) to communicate with the reader. 

Otherwise, I would recommend using standard Javascript keyboard event  APIs to read from the "keyboard" that is the RFID reader. 

Closing this as this isn't a node-hid issue.  Feel free to keep updating this issue, but it's not a node-hid issue.

closed 2023-05-17 by todbot

echtme, 2023-05-18:
Hey todbot

Update: The latest Electron version now allows me to create the app, and it's working sometimes. However, I occasionally encounter this issue (see below) and cannot reproduce the error.

Can we resetting the ports before accessing them? Would this potentially solve the problem?

Script execution stderr: /Users/me/Development/project/node_modules/node-hid/nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

TypeError: cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS08@14300000/USB Reader@14300000/Keyboard@0/AppleUserUSBHostHIDDevice
    at new HID (/Users/me/Development/project/node_modules/node-hid/nodehid.js:49:17)
    at /Users/me/Development/project/reader.js:25:20
    at Array.forEach (<anonymous>)
    at Object.<anonymous> (/Users/me/Development/project/reader.js:23:15)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1157:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:77:12)
    at node:internal/main/run_main_module:17:47
    
    reader.js (25): const device = new HID.HID(rfidReader.path);

todbot, 2023-05-18:
If your RFID reader is acting like a keyboard, you are fighting the OS for ownership of the device. node-hid cannot help with this.