node-hid/node-hid #512: Got error when using `setNonBlocking` function

SpiritLooper, 2023-05-27:
I want to set "non blocking mode" while reading data from my BuzzQuizz Wireless Controller and i have this error when i call the function `hid.setNonBlocking(true)` : 

```node
Error: Error in native callback
    at Object.<anonymous> (C:\Users\SpiritLooper\Documents\code\QuizzPartyGame\server\buzzerlib.js:8:5)    
    at Module._compile (node:internal/modules/cjs/loader:1254:14)
    at Module.load (node:internal/modules/cjs/loader:1117:32)
    at Module._load (node:internal/modules/cjs/loader:958:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:23:47

Node.js v18.16.0
```

This function work in version `1.3.2` but doesn't in the latest release (`2.1.2`).
I'm working on Windows 10 22H2

Julusian, 2023-05-27:
could you try doing `hid.setNonBlocking(1)`? It looks like it is expecting a numeric parameter

SpiritLooper, 2023-05-28:
> could you try doing `hid.setNonBlocking(1)`? It looks like it is expecting a numeric parameter

That's work ! Maybe we could [update the README file](https://github.com/node-hid/node-hid#devicesetnonblockingno_block) ?