node-hid/node-hid #517: Wrong buffer length when using getFeatureReport

Mola19, 2023-08-08:
The Buffer i get when calling `getFeatureReport()` has one more element at the end than it should

`console.log(dev.getFeatureReport(0x04, 4)`
output: `[ 4, 1, 3, 1, 84 ]`

Wireshark also shows that the number isn't being sent via usb 
[image]

The number also changes every time and seems very random

I am pretty sure that the issue is, that the code doesn't account for the report id and reads 4 bytes, although it should only read 3, this means it will just read into random memory

todbot, 2023-08-08:
Correct. If your device uses reportIds, then the length passed to `getFeatureReport()` needs to be one more than the report size.

Mola19, 2023-08-08:
I am not sure what you mean?
If add one to the report i get another byte of garbage memory.
The issue isn't that i want a longer output, i want a shorter output, but i can't input 3 instead of 4 as the length because it then can't read anymore and errors, as the report size is 4.

Mola19, 2023-08-08:
> If i add one to the report i get another byte of garbage memory.
There is even a difference, because that on gets send by the device, the one i mean comes from system memory.

I might be wrong, but for me it seems like there is an issue in node-hid or hidapi, which makes it read to far into system memory

todbot, 2023-08-08:
You are not asking for enough bytes for your report. You should ask for 5 bytes.  node-hid/hidapi is giving you back the correct size buffer, since it's abiding by the HID Report Descriptor, but the OS you're on (MacOS?) is only giving the 3 of your device's 4 report bytes.

If your devices HID Report Descriptor says that if the specified Feature Report's REPORT_SIZE is 8 and REPORT_COUNT is 4 and REPORT_ID is set, the data sent and received to the device will be 5 bytes long

Mola19, 2023-08-08:
Oh you mean the code on the device is wrong?
Ok then i just have to ignore it, since it is proprietary firmware

Mola19, 2023-08-10:
But if i do this then i get `[ 4, 1, 3, 1, 255, 3 ]. The output that i want (which is also what is actually sent via usb) is `[ 4, 1, 3, 1 ]`. Whatever length i give to getFeatureReport, it will give me one more byte at the end, which is not sent over usb and shouldn't be there

Julusian, 2023-08-10:
It looks like you might be hitting https://github.com/libusb/hidapi/pull/232.
This project is currently using a version of hidapi that predates that fix.

I have tried to read the code and checked the docs for the hidapi method, and we are doing exactly as we are supposed to, returning the number of bytes that hidapi told us were read.

Mola19, 2023-08-13:
Yes i can confirm that using hidapi 0.14.0 instead of 0.10.1 will fix the issue.

Should i close this issue then?

closed 2025-06-03 by Julusian