node-hid/node-hid #520: I get an error when I unplug my device

roman-16, 2023-10-24:
When I build a connection to my device and then unplug it, I get a dialog with this error: `could not read from HID device`

I would be easier for me if I get no dialog at all.

Julusian, 2023-10-24:
Yeah, this happens but there is no 'close' event currently, so this is the only way to know that a device is no longer available for use.
It could be nice to tidy that up and add a new event for device going away, but that would be a breaking change and would need some investigation to see what errors could be left

roman-16, 2023-10-24:
I understand, thank you. Would be great if this can be implemented in the future :slightly_smiling_face:

todbot, 2023-10-24:
In `hidapi` (the library that `node-hid` wraps), there isn't (yet) an ability to do device plug/unplug detection, so you either check the state of your read/write calls and catch the errors indicating the device went away or some folks have used `node-usb-detection` to do general USB device plug/unplug events. (this only works for USB though, not Bluetooth)

roman-16, 2023-10-24:
Got it. Sadly libraries like `node-usb-detection` fire after the error dialog of `node-hid` has been opened.

todbot, 2023-10-24:
Yeah this is why I typically wrap my reads & writes in `try { } catch {}` blocks to decide programmatically what to do when I/O fails. (as one would do when reading/writing to disk or network)