node-hid/node-hid #522: How to get a device type

skumarverificient, 2023-11-21:
var HID = require('node-hid');
    const devices = HID.devices()
    console.log("devices",devices)

todbot, 2023-11-21:
What kind of "type" do you mean?

skumarverificient, 2023-11-24:
device like keyboard,mouse headphone etc..

todbot, 2023-11-24:
For types of HID devices (like mice and and keyboards), you want to look at the `usagePage` and `usage` properties returned from `HID.devices()`.  You can get a list of HID Usages in this document: https://usb.org/sites/default/files/hut1_4.pdf

Note, while you can see keyboard and mouse with `node-hid`, you cannot access them as they're owned by the OS.  Also, headphones are audio devices. 

If you're asking about general USB device class (e.g. HID vs Audio vs Mass Storage vs CDC/Serial), then you cannot use  `node-hid` but instead should look at `node-usb`. The defined USB device codes are listed here: https://www.usb.org/defined-class-codes

closed 2023-11-24 by todbot

skumarverificient, 2023-12-01:
On the Windows machine, I have connected a USB mouse and a USB keyboard. I'm trying to filter out the names of the mouse and keyboard from the given output. I've noticed that several objects have a usage of 1. For your information, the mouse is named 'Dell MS116 USB Optical Mouse' and the keyboard is named 'USB Keyboard'.       
   const devices = HID.devices()
   console.log("devices",devices )
   
   devices [
  {
    vendorId: 1102,
    productId: 4639,
    path: '\\\\?\\hid#dell0816&col05#5&2c0c5c6d&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: '9999',
    manufacturer: 'Microsoft',
    product: 'HIDI2C Device',
    release: 560,
    interface: -1,
    usagePage: 13,
    usage: 14
  },
  {
    vendorId: 32903,
    productId: 2590,
    path: '\\\\?\\hid#intc816&col01#3&36a7043c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    release: 512,
    interface: -1,
    usagePage: 1,
    usage: 12
  },
  {
    vendorId: 32903,
    productId: 2590,
    path: '\\\\?\\hid#intc816&col02#3&36a7043c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    release: 512,
    interface: -1,
    usagePage: 1,
    usage: 13
  },
  {
    vendorId: 1102,
    productId: 4626,
    path: '\\\\?\\hid#vid_044e&pid_1212&col01&col02#7&1d3f22cb&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',
    serialNumber: '',
    manufacturer: '',
    product: '',
    release: 46873,
    interface: -1,
    usagePage: 1,
    usage: 6
  },
  {
    vendorId: 12538,
    productId: 8241,
    path: '\\\\?\\hid#vid_30fa&pid_2031&mi_01&col04#7&e07fd06&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',
    manufacturer: 'INSTANT',
    product: 'USB Keyboard',
    release: 256,
    interface: 1,
    usagePage: 1,
    usage: 6
  },
  {
    vendorId: 12538,
    productId: 8241,
    path: '\\\\?\\hid#vid_30fa&pid_2031&mi_01&col06#7&e07fd06&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'INSTANT',
    product: 'USB Keyboard',
    release: 256,
    interface: 1,
    usagePage: 65280,
    usage: 1
  },
  {
[... 161 lines cut ...]
    usagePage: 65281,
    usage: 1
  },
  {
    vendorId: 1118,
    productId: 0,
    path: '\\\\?\\hid#converteddevice&col01#5&379854aa&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',
    release: 0,
    interface: -1,
    usagePage: 1,
    usage: 6
  },
  {
    vendorId: 1102,
    productId: 4639,
    path: '\\\\?\\hid#dell0816&col04#5&2c0c5c6d&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: '9999',
    manufacturer: 'Microsoft',
    product: 'HIDI2C Device',
    release: 560,
    interface: -1,
    usagePage: 65282,
    usage: 1
  }
]

skumarverificient, 2023-12-07:
Thank you for your quick response.
Can you help me to filter out keyboard and mouse from HID.devices()

Here i am not getting correct result
For mouse
 const usbMouse = devices.filter((el) => {
        if (el.usagePage === 1 && device.usage === 2) {
            return true
        }
    });
    
For Keyboard
const usbKeyboard = devices.filter((el) => {
        if (el.usagePage === 1 && device.usage === 1) {
            return true
        }
    });

todbot, 2023-12-07:
What is the result you expect and what is the result you are getting? (and please format your code / output to make it easier to read)

skumarverificient, 2023-12-13:
I'm encountering an issue while attempting to distinguish between a USB mouse and keyboard using JavaScript. I've applied filters based on the usagePage and usage properties. However, in the case of the USB dongle, both the mouse and keyboard exhibit identical usagePage and usage values. Consequently, the device named '2.4G Receiver' appears in both the mouse and keyboard lists, whereas ideally, it should only belong to the usbMouse list.

const usbMouse = devices.filter((el) => {
    if (el.usagePage === 1 && el.usage === 2) {
        return true;
    }
});

const usbKeyboard = devices.filter((el) => {
    if (el.usagePage === 1 && el.usage === 6) {
        return true;
    }
});

The output showcases the following:
usbMouse: [
  {
    vendorId: 9639,
    productId: 64035,
    path: '\\\\?\\hid#vid_25a7&pid_fa23&mi_00#7&102e827e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',
    manufacturer: 'Compx',
    product: '2.4G Receiver',
    release: 577,
    interface: 0,
    usagePage: 1,
    usage: 6
  }
]

usbKeyboard: [
  {
    vendorId: 9639,
    productId: 64035,
    path: '\\\\?\\hid#vid_25a7&pid_fa23&mi_00#7&102e827e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',
    manufacturer: 'Compx',
    product: '2.4G Receiver',
    release: 577,
    interface: 0,
    usagePage: 1,
    usage: 6
  }
]

todbot, 2023-12-13:
This is the problem with usagePage=1 (and really most devices with usagePages owned by the OS), you won't get reasonable results because either the OS lies, the devices lie (to work with the OS better), or it's complicated because many devices are both mouse and keyboards (e.g. keyboards with built-in trackpads, mice with a dozen buttons).

It doesn't matter anyway, since `node-hid` and `hidapi` cannot access usagePage=1 devices anyway.

What is it you're trying to accomplish?

skumarverificient, 2023-12-14:
I need to display two separate lists of the mouse and keyboards connected to the user's machine

todbot, 2023-12-14:
For compound devices like that wireless receiver, I don't think you'll be able to distinguish.

skumarverificient, 2023-12-14:
Yes, I am only encountering issues with the wireless receiver. Are there any workarounds or alternative approaches available? Any help would be greatly appreciated.