node-hid/node-hid #526: Something wrong with pre-built binaries selection at 3.x

Sfinx, 2023-12-15:
I'm using node-hid at electron and building win32 at Linux. Noticed that starting from node-hid v3.x the HID.node module is always Linux ELF binary when buildingfor win32 but it must be PE32+ one. Just changing the node-hid version back to v2.x fixed the things. v2.x build log:

```
 App •  WAIT  • Bundling app with electron-builder...
  • electron-builder  version=24.9.1 os=6.6.5-060605-generic
  • writing effective config  file=dist/electron/Packaged/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=node-hid@2.2.0 platform=win32 arch=x64
  • install prebuilt binary  name=node-hid version=2.2.0 platform=win32 arch=x64 napi=
• packaging       platform=win32 arch=x64 electron=28.0.0 appOutDir=dist/electron/Packaged/win-unpacked
```

v3.x build log:

``` App •  WAIT  • Bundling app with electron-builder...
  • electron-builder  version=24.9.1 os=6.6.5-060605-generic
  • writing effective config  file=dist/electron/Packaged/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=node-hid@3.0.0 platform=win32 arch=x64
```

Despite it says that native dependencies are rebuilding for platform=win32 arch=x64 - the Linux ELF is built always. Not sure if this is electron-builder bug but changing the single node-hid line in package.json to ^2.2.0 fixes this pre-built selection logic bug for me.

Julusian, 2023-12-15:
One solution is to set `npmRebuild: false` in your electron-builder config. you dont appear to have any other native dependencies, and `node-hid` is unlikely to need recompiling (unless you are building for an exotic os)

I shall take a look at whether there are any solutions that can be done on this side of things. It really needs to not be recompiling node-hid, as trying to build for another os is highly unlikely to work

Sfinx, 2023-12-15:
`npmRebuild: false` is working ! The logs now is:

```
 App •  WAIT  • Bundling app with electron-builder...
  • electron-builder  version=24.9.1 os=6.6.5-060605-generic
  • writing effective config  file=dist/electron/Packaged/builder-effective-config.yaml
  • skipped dependencies rebuild  reason=npmRebuild is set to false
  • packaging       platform=win32 arch=x64 electron=28.0.0 appOutDir=dist/electron/Packaged/win-unpacked
```

The pre-built binary is now correctly selected and is working. Thanks !

NB: No exotic OS, it is just amd64 Ubuntu where the boring win32 electron app is building.

closed 2023-12-15 by Sfinx