node-hid/node-hid #529: pre-builds of 3.0.0 are not compatible with electron-builder

ert78gb, 2024-01-14:
Hi All,

I am upgraded `node-hid` from `2.2.0` to `3.0.0` version. I had to modify the CI setup to build node-hid from source because the pre-builds are not compatible.

- PR https://github.com/UltimateHackingKeyboard/agent/pull/2162
- node version: `18.17.1`
- electron version: `27.2.2`
- electron-builder version: `24.6.4`. It is not the latest, but upgrading to the latest does not solve the problem. Build: https://github.com/UltimateHackingKeyboard/agent/actions/runs/7519252321?pr=2162

I also recognised the `binary.napi_versions` is `[4]` in the [package.json](https://github.com/node-hid/node-hid/blob/484d13460a89939530eb9e562d5c5f8aa2e56dc5/package.json#L33C6-L33C19), but `napi_versions` is `[3]` in the [binding-options.js](https://github.com/node-hid/node-hid/blob/484d13460a89939530eb9e562d5c5f8aa2e56dc5/binding-options.js#L3C5-L3C23). Should not be the 2 config same?

Julusian, 2024-01-14:
Something probably needs adding to the readme for this..

I think this is the same issue as https://github.com/node-hid/node-hid/issues/526.

The problem is that there isn't a way (that I've found) to tell electron-builder that this addon has prebuilds and doesn't need to be recompiled. So it decides that as this is a native module it must need recompiling, and is then failing as you haven't installed the dependencies necessary for that to work.
It used to work because node-hid used to use a different library for prebuilds that electron-builder had a special case in their code to handle.

If you don't have any other native dependencies, you can add `npmRebuild: false` in your electron-builder config. Or it might be possible to tell electron-builder which dependencies to rebuild/skip. 
Another option would be to install the dependencie it needs, but that could reduce the platform support of the produced binaries (the prebuilds are built to have the widest support possible, which requires additional setup and careful choice of os versions)

> I also recognised the binary.napi_versions is [4] in the [package.json](https://github.com/node-hid/node-hid/blob/484d13460a89939530eb9e562d5c5f8aa2e56dc5/package.json#L33C6-L33C19), but [...]

Yeah, they should be the same. I don't think anything uses the one in the package.json though, I am pretty sure it's presence is used by some tool to check if the library is a node-api based native library and doesn't care about the value beyond that

ert78gb, 2024-01-15:
@Julusian thx for the fast response. I set `npmRebuild: false` and works on all OS that available for me. I am waiting for the users feedback. I am closing this issues and I will reopen it if necessary.

closed 2024-01-15 by ert78gb