node-hid/node-hid #532: Segmentation Fault on both Deno and Bun

Repugraf, 2024-02-02:
I suppose this is the issue with deno and bun. But could you share insight why it's failing for them?
Does it have to do something with Rust and Zig?

It does fails only if you attach on data listener:

```js
const device = new HID(vendorId, productId); // everything is fine

/**
 * DENO:
 * zsh: segmentation fault  deno run -A test.mjs
 *
 * BUN:
 * Message received: Buffer(19) [ 0, 0, 0, 0, 4, 0, 4, 0, 0, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0 ]
 * Message received: Buffer(19) [ 0, 0, 0, 0, 4, 0, 4, 0, 0, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0 ]
 * zsh: segmentation fault  bun test.mjs
 *
 */
device.on("data", buffer => console.log("Message received:", buffer));
```

Device tested: MacOS M1 chip

Julusian, 2024-02-02:
Could be hitting https://github.com/denoland/deno/issues/21376

And bun doesn't have full node-api coverage https://github.com/oven-sh/bun/issues/158, so it is expected to not work there. From a quick skim we are using `napi_open_escapable_handle_scope`, possibly other methods they havent implemented

Repugraf, 2024-08-28:
Just tested with deno `1.46.1` and it works. Also it seems that the issue was fixed https://github.com/denoland/deno/issues/21376

Repugraf, 2024-08-28:
Just tested bun `1.1.26` and it works. Issue mentioned by @Julusian is not fixed but probably another issue causing seg fault was fixed

Repugraf, 2024-08-28:
It seems that both deno and bun fixed the issue on their side and it looks like this issue can be closed.

closed 2024-08-28 by Repugraf