node-hid/node-hid #537: [crash] the app when remove and close the device listeners when using async node-hid approach

oalfroukh, 2024-03-07:
Hi,
- I'm using `HID` and I would like to move to use `HIDAsync`, **BUT** when the user selects another headset from our app, we will remove the hid device `listeners` and `close` the hid, but on that time the app will `crash`

- `node-hid` version is `3.0.0`

- Here is [Fiddle project](https://gist.github.com/035e452433ccf63598a1a97a489188f8) to reproduce the crash, 

so to reproduce the issue replace the `PRODUCT_ID` and `VENDOR_ID` with your headset values and run the project on your machine using `electron Fiddle` app

[image]

Julusian, 2024-03-07:
This sounds like it could be fixed by #533
You can try the fix by using `@julusian/hid` v3.0.1-0

oalfroukh, 2024-03-07:
> This sounds like it could be fixed by #533 You can try the fix by using `@julusian/hid` v3.0.1-0

YES! It seems working fine, Thanks! 
When do you plan to provide that fix on this package, because our app will be in production in the next period and this crash is very critical for us.