node-hid/node-hid #538: crash electron app when using node-hid

oalfroukh, 2024-03-24:
Hi,

- I'm facing this crash on our app when using `node-hid` `v3.00` or `@julusian/hid` `v3.0.1-0` (sync or async methods) and I would say it's consistent (1 in 4), I couldn't create a test case via a quick start electron app to use it from your side as our app is complex a little bit. 

- But, I see these logs on the main process when crash the app (I hope this stack trace will help you on the investigation):

```
C:\Windows\system32\cmd.exe [35192]: ..\..\third_party\electron_node\src\api\callback.cc:74: Assertion `(Environment::GetCurrent(isolate)) == (env)' failed.
 1: 00007FF7A3028506 node::SetTracingController+86134
 2: 00007FF7A30280DC node::SetTracingController+85068
 3: 00007FF7A30A6086 node::CallbackScope::CallbackScope+502     
 4: 00007FF7A30A5ECD node::CallbackScope::CallbackScope+61      
 5: 00007FF7A3F2DBE4 node::AsyncResource::CallbackScope::CallbackScope+84
 6: 00007FF7A37F959D node_api_get_module_file_name+1885
 7: 00007FF7A370CA3C uv_dlerror+71276
 8: 00007FF7A370C893 uv_dlerror+70851
 9: 00007FF7A381FA34 uv_fs_poll_getpath+1444
10: 00007FF7A30AB2D0 uv_run+1888
11: 00007FF7A30AAD6B uv_run+507
12: 00007FF7A307741C node::loader::ModuleWrap::Initialize+164380
13: 00007FF7A3077D42 node::loader::ModuleWrap::Initialize+166722
14: 00007FF7A30A340A node::FreeEnvironment+186
15: 00007FF7A11DFE95 v8::Extension::auto_enable+981
16: 00007FF7A11FFB29 v8::Extension::auto_enable+131177
17: 00007FF7A2EF572F uv_fs_get_system_error+1556319
18: 00007FF7A418B607 uv_version_string+2480503
19: 00007FF7A3B3A25D uv_stream_set_blocking+3246621
20: 00007FF7A343951D GetHandleVerifier+3530541
21: 00007FF7A3507E53 GetHandleVerifier+4376675
22: 00007FF7A3508A89 GetHandleVerifier+4379801
23: 00007FF7A34ADFDE GetHandleVerifier+4008430
24: 00007FF7A34B1563 GetHandleVerifier+4022131
25: 00007FF7A2E1A7FB uv_fs_get_system_error+659499
26: 00007FF7A2EEA529 uv_fs_get_system_error+1510745
27: 00007FF7A2F000D3 uv_fs_get_system_error+1599747
28: 00007FF7A2EFFE40 uv_fs_get_system_error+1599088
29: 00007FF7A2EE8F86 uv_fs_get_system_error+1505206
30: 00007FF7A361EFF9 GetHandleVerifier+5519881
31: 00007FF7A1A39319 v8::StartupData::operator=+666009
32: 00007FF7A361F85A GetHandleVerifier+5522026
33: 00007FF7A2AB660B IsSandboxedProcess+23995
34: 00007FF7A52F13A7 Cr_z_adler32+728823
35: 00007FF7A5252BCB Cr_z_adler32+79643
36: 00007FF7A5271B5E Cr_z_adler32+206510
37: 00007FF7A521180B cppgc::internal::WriteBarrier::DijkstraMarkingBarrierRangeSlow+4512667
38: 00007FF7A52D7757 Cr_z_adler32+623271
39: 00007FF7A52DFD03 Cr_z_adler32+657491
40: 00007FF7A30EA5A9 GetHandleVerifier+61369
41: 00007FF7A2A0FC48 uv_os_getpid+181320
42: 00007FF7A2EE12E0 uv_fs_get_system_error+1473296
43: 00007FF7A12ED862 v8::internal::StrongRootAllocatorBase::StrongRootAllocatorBase+542194
44: 00007FF7A12EE6D9 v8::internal::StrongRootAllocatorBase::StrongRootAllocatorBase+545897
45: 00007FF7A12EA9E0 v8::internal::StrongRootAllocatorBase::StrongRootAllocatorBase+530288
46: 00007FF7A12EAB5D v8::internal::StrongRootAllocatorBase::StrongRootAllocatorBase+530669
47: 00007FF7A102CBC5 std::__Cr::vector<v8::CpuProfileDeoptInfo,std::__Cr::allocator<v8::CpuProfileDeoptInfo> >::vector<v8::CpuProfileDeoptInfo,std::__Cr::allocator<v8::CpuProfileDeoptInfo> >+60933
48: 00007FF7A56C8642 Cr_z_adler32+4755858
49: 00007FFFCB72257D BaseThreadInitThunk+29
50: 00007FFFCC6CAA58 RtlUserThreadStart+40
Renderer process crashed - see https://www.electronjs.org/docs/tutorial/application-debugging for potential debugging information.
```

- Also I see on our app logs that a crash happens after calling `await this.managedDevice.write(payload)` and `payload` is number[]  and FYI `write` doesn't resolve/reject on the `async` approach (of course there are no errors with `sync` approach while the crash only)

- I see there is a report on sentry related to `node-hid` but I'm not sure if it's the same use case I reproduce, here is the stack trace if it will help:

```
OS Version: Windows 10.0.22631 (3296)
Report Version: 104

Crashed Thread: 13096
[... 341 lines cut ...]
Thread 1696 name: AudioOutputDevice
0   ntdll.dll                       0x7fffcc70fec4      NtWaitForMultipleObjects
1   KERNELBASE.dll                  0x7fffc9fd0478      WaitForMultipleObjectsEx
2   KERNELBASE.dll                  0x7fffc9fd037d      WaitForMultipleObjects
3   electron.exe                     0x7ff7bba0015e      base::`anonymous namespace'::CancelableFileOperation<T> (sync_socket_win.cc:156)
4   electron.exe                     0x7ff7bb9fffb9      base::CancelableSyncSocket::Receive (sync_socket_win.cc:296)
5   electron.exe                     0x7ff7ba5d989d      media::AudioDeviceThread::ThreadMain (audio_device_thread.cc:85)
6   electron.exe                     0x7ff7bb9fe7b3      base::`anonymous namespace'::ThreadFunc (platform_thread_win.cc:140)
7   KERNEL32.DLL                    0x7fffcb72257c      BaseThreadInitThunk
8   ntdll.dll                       0x7fffcc6caa57      RtlUserThreadStart

Thread 7264
0   ntdll.dll                       0x7fffcc70f3f4      NtWaitForSingleObject
1   KERNELBASE.dll                  0x7fffc9fa497d      WaitForSingleObjectEx
2   HID.node                        0x7fff3cd6c778      <unknown>
3   HID.node                        0x7fff3cd675a2      <unknown>
4   HID.node                        0x7fff3cda149e      <unknown>
5   KERNEL32.DLL                    0x7fffcb72257c      BaseThreadInitThunk
6   ntdll.dll                       0x7fffcc6caa57      RtlUserThreadStart

EOF

```

Thanks For the support

Julusian, 2024-03-25:
I don't think I have tried this in electron myself in a while.
Could this be happening when the hid device is disconnected/closed?

Does it behave any differently if you pass an Buffer instead of an array? You could do `Buffer.from(arr)` to convert your array

oalfroukh, 2024-03-26:
> Buffer.from(arr)

Hi @Julusian 
Thanks for the response!
the crash is reproducible also when using Buffer. I just used `await managedDevice.write(Buffer.from([11,0,0]));`

the scenario isn't related to disconnecting/closing the hid device, it's when trying to release the outgoing call to the headset

oalfroukh, 2024-03-31:
Hi @Julusian 
- I figured out new info about the case, it seems the crash is reproducible only when there is a child window that's created by `window.open()`. I found reported issues on Eeletron about this case but I'm not 100% sure if it's the same root cause or not https://github.com/electron/electron/issues/36858 and https://github.com/electron/electron/issues/33868
- What I don't understand on my side is that I don't use `node-hid` on that child window the same as the other reported users on the above `issues` (they used native libraries on the child windows)