node-hid/node-hid #541: Throwing in event listeners crashes the process and does not print the exception

alvaro-cuesta, 2024-05-05:
What the title says.

Reproducing code:

```
const QMK_VID = 0x3434
const QMK_PID = 0x0363
const QMK_RAW_USAGE_PAGE = 0xff60
const QMK_RAW_USAGE_ID = 0x61

devicesAsync()
  .then(devices => {
    const device = devices.find(device =>
      device.vendorId === QMK_VID
      && device.productId === QMK_PID
      && device.usagePage === QMK_RAW_USAGE_PAGE
      && device.usage === QMK_RAW_USAGE_ID
    )

    console.log(device)

    if (!device) {
      console.error('QMK raw device not found')
      return
    }

    return HIDAsync.open(device.path)
  })
  .then(hid => {
    console.log('Listening for data...')

    hid.on('data', data => {
      throw new Error('Foobar')
    })
  })
```

As soon the `throw` is reached, the process exits completely and the exception is not printed anywhere.

Hate9, 2024-06-10:
#543 does not fix the issue and instead causes the program to throw a *different* error:
```
[...]/node_modules/node-hid/nodehid.js:122
                    this.emit("error", e);
                         ^

TypeError: this.emit is not a function
    at Immediate.<anonymous> ([...]/node_modules/node-hid/nodehid.js:122:26
```

Julusian, 2024-06-10:
Ah.. I did test this with the HIDAsync class, and didn't think I needed to test HID too. But HID is written in an older style of js and needs a this to be replaced with self

PR #544 (2024-10-21): fix: typo in error handling of HID
I made a mistake in #543 (to fix #541) for the HID class, which causes errors to be misthrown when emitted by user code
closed 2024-10-21 by todbot