node-hid/node-hid #548: Add deno compatiblity

hsk81, 2024-10-03:
Using `node-hid` inside compiled deno projects does not seem to work. However, using `node-hid` inside a project does seem to work. It boils down to path resolution (I think):

**Working example,** where the `--ledger` option causes the (indirect) use of `node-hid`:
```sh
deno -A ./cli/banq/banq-main.ts refresh xpow apow --broadcast --ledger ## no errors are thrown
```

**Failing example,** where the `--ledger` option causes the (indirect) search attempt of `node-hid`:
```sh
deno task build-banq-linux
```
```sh
./dist/banq.x86_64-linux.run refresh xpow apow --broadcast --ledger
```
```
error: Uncaught (in promise) TypeError: /tmp/deno-compile-banq.x86_64-linux.run/banq-cli.git/node_modules/.deno/usb@2.9.0/node_modules/usb/prebuilds/linux-x64/node.napi.glibc.node: cannot open shared object file: No such file or directory
    at Object.Module._extensions..node (node:module:790:21)
    at Module.load (node:module:655:32)
    at Function.Module._load (node:module:523:13)
    at Module.require (node:module:674:19)
    at require (node:module:801:16)
    at load (file:///tmp/deno-compile-banq.x86_64-linux.run/banq-cli.git/node_modules/.deno/node-gyp-build@4.8.2/node_modules/node-gyp-build/node-gyp-build.js:22:10)
    at Object.<anonymous> (file:///tmp/deno-compile-banq.x86_64-linux.run/banq-cli.git/node_modules/.deno/usb@2.9.0/node_modules/usb/dist/usb/bindings.js:8:36)
    at Object.<anonymous> (file:///tmp/deno-compile-banq.x86_64-linux.run/banq-cli.git/node_modules/.deno/usb@2.9.0/node_modules/usb/dist/usb/bindings.js:11:4)
    at Module._compile (node:module:736:34)
    at Object.Module._extensions..js (node:module:757:11)
```

The issue seems to be that the compiled deno distributeable of `./dist/banq.x86_64-linux.run` is unpacked at runtime under `/tmp` messing up the resolution for `node-hid`. Reported this also in the `deno` forums, but was told that compiled deno distributables don't support *native* Node modules yet.. perhaps, there is a work-around?

Manually fixing the path by symbolic linking *or* copying the required builds fails also with `undefined symbol: napi_module_register`.

hsk81, 2024-10-03:
BTW: The corresponding Github repository is at
> https://github.com/blackhan-software/xpower-banq-cli

Julusian, 2024-10-03:
I would guess that this is a problem of the dynamic require we do for `node.napi.glibc.node` is missing because deno couldn't statically determine the file was needed, so it didn't include it.
Fixing that on our side would mean rewriting the code that does that require to use a static list of expected filenames, which could well cause other issues.

> Manually fixing the path by symbolic linking or copying the required builds fails also with undefined symbol: napi_module_register.

There is nothing we can do about that, `napi_module_register` is a core part of the api that this module expects to be able to invoke, it missing suggests `deno distributables don't support native Node modules yet`. There isn't anything we can do about that.

hsk81, 2024-10-03:
Thanks for the answer! The only approach that worked so far, was to publish my package to deno's JSR registry, and then *manually* copy `node-hid/build` from the projects local `node_modules` to deno's cache:

```
deno run -A jsr:@blackhan/xpower-banq-cli refresh xpow apow --broadcast --ledger
```
```
error: Uncaught (in promise) Error: Could not locate the bindings file. Tried:
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/build/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/build/Debug/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/build/Release/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/out/Debug/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/Debug/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/out/Release/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/Release/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/build/default/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/compiled/20.11.1/linux/x64/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/addon-build/release/install-root/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/addon-build/debug/install-root/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/addon-build/default/install-root/HID_hidraw.node
 → /home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/lib/binding/node-v108-linux-x64/HID_hidraw.node
    at bindings (file:///home/tux/.cache/deno/npm/registry.npmjs.org/bindings/1.5.0/bindings.js:126:9)
    at loadBinding (file:///home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/nodehid.js:19:46)
    at Object.showdevices (file:///home/tux/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/nodehid.js:126:5)
    at getDevices (file:///home/tux/.cache/deno/npm/registry.npmjs.org/@ledgerhq/hw-transport-node-hid-noevents/6.30.5/lib/TransportNodeHid.js:26:31)
    at Function.TransportNodeHid.listen (file:///home/tux/.cache/deno/npm/registry.npmjs.org/@ledgerhq/hw-transport-node-hid/6.29.5/lib/TransportNodeHid.js:89:69)
    at file:///home/tux/.cache/deno/npm/registry.npmjs.org/@ledgerhq/hw-transport/6.31.4/lib/Transport.js:178:30
    at new Promise (<anonymous>)
    at Function.create (file:///home/tux/.cache/deno/npm/registry.npmjs.org/@ledgerhq/hw-transport/6.31.4/lib/Transport.js:176:16)
    at new LedgerSigner (https://jsr.io/@blackhan/xpower-banq-cli/0.1.7/wallet/ledger-signer.ts:63:29)
    at wallet (https://jsr.io/@blackhan/xpower-banq-cli/0.1.7/wallet/ledger.ts:36:18)
```

AFTER copying the `node-hid/build` to deno's cache I can at least use `deno run` (bypassing the compiled executable):

```
cp -r ./node_modules/node-hid/build/ ~/.cache/deno/npm/registry.npmjs.org/node-hid/2.1.2/
```